ros2/ros2 #1: merge api_review branches

wjwwood, 2015-03-03:
There are `api_review` branches on several repositories:

- `examples`
- `ros_middleware_interface`
- `ros_middleware_opensplice`
- `rosidl`
- `rosidl_dds`

In addition to these branches there are some new repositories with new packages:

- `rcl`
- `rclc`

These branches and new repositories mainly refactor the major internal APIs (`ros_middleware_interface` and `rcl`) to be C only rather than C++ and makes some other structural changes to better support C and C++ at the same time.

These branches are in a partial working state. There are some things that work:

- C and C++ code `.msg` generation
- `ros_middleware_opensplice` has been updated to use new C only `ros_middleware_interface`

Other things do not work:

- `rcl` and `rclc` are only headers, they have no implementation.
- `rclcpp` does not use `rcl`.
- The two Connext implementations have not been updated at all.
- The new service work @esteve did is not integrated yet into the new C-only version of the ros_middleware_interface headers.

wjwwood, 2015-03-04:
I created `pre_api_review` branches on the repositories which needed it. We should remove those when we no longer need them.

wjwwood, 2015-03-04:
I merged all of the branches to master, but it's still in a pretty broken state. I think we will have to resolve the Service implementation before we call this issue done since they were already on master.

wjwwood, 2015-03-05:
Master is compiling for me, the talker and listener example works, and I've added the new repositories to base setup.

Can I get reviews, +1's, for closing this?

wjwwood, 2015-03-05:
@dirk-thomas were you able to compile and run the basics (satisfying this issue)?

@esteve @tfoote can you try this out, I'd like to close the issue.

dirk-thomas, 2015-03-05:
+1

I was able to `build` the current state and only ran `publisher` / `subscriber` which was working.

closed 2015-03-05 by wjwwood