ros2/ros2 #2: Setup batch CI for Linux

wjwwood, 2015-03-03:
Acceptance Criteria: 

- [ ] Create a batch script type CI testing to be batch run
- [ ] report test result in jenkins job (# passed/failed)

tfoote, 2015-03-07:
Basic scripts working here: https://github.com/tfoote/ros2_batch_ci
Jenkins job on the test farm here: http://54.183.26.131:8080/job/ros2_batch_ci/

Just need to switch back to master from pre_api_review branches.

dirk-thomas, 2015-03-07:
Permission / ownership problem prevents to read the test results in the workspace: see http://54.183.26.131:8080/job/ros2_batch_ci/ws/ros2_ws/build/osrf_pycommon/

tfoote, 2015-03-08:
@dirk-thomas The scripts are in a repo here for now: filed as in issue: https://github.com/tfoote/ros2_batch_ci/issues/3

tfoote, 2015-03-09:
Job setup [here](http://54.183.26.131:8080/job/ros2_batch_ci/), and collecting test results. [ci job source](https://github.com/tfoote/ros2_batch_ci)

closed 2015-03-10 by tfoote