ros2/ros2 #3: Update packages to use QA tools

wjwwood, 2015-03-03:
AC:

- [ ] Example packages running QA tools like linters
- [ ] documentation about how to use tools
- [ ] update existing packages to use the existing tools, as an automated test

----

dirk-thomas, 2015-03-06:
ament/ament_lint#2 implements the necessary CMake infrastructure to register default linters with minimal code.

dirk-thomas, 2015-03-06:
ros2/rosidl#5 updates two packages to use the new functionality as well as register its nosetests.

wjwwood, 2015-03-06:
Connects to ros2/rosidl#5

dirk-thomas, 2015-03-07:
The linter packages now also use other linters as far as possible: https://github.com/ament/ament_lint/commit/9c9fd1e06f950ad3f1a235f029b35b112b9d1ca2

dirk-thomas, 2015-03-12:
Do you consider this (https://github.com/ament/ament_lint/blob/docs/ament_clang_format/doc/index.rst) to be sufficient to describe how to use the linters? I just want to get some first feedback before writing N if these.

dirk-thomas, 2015-03-12:
The configuration for `cppcheck` is pretty standard and looks reasonable to me. It currently does not generate a failing test.

dirk-thomas, 2015-03-12:
The documentation for the linter packages is in: ament/ament_lint#9

closed 2015-03-13 by dirk-thomas