ros2/ros2 #6: Setup Batch CI for Windows

wjwwood, 2015-03-03:


tfoote, 2015-03-10:
Current test environment Windows 8.1 Visual Studion 2013

tfoote, 2015-03-13:
It looks like we might be better off trying out appveyor. I'm experimenting here: https://ci.appveyor.com/project/tfoote/ros2-batch-ci

tfoote, 2015-03-14:
Notes on setting up the slave: https://gist.github.com/tfoote/caaa9a5c31246ec5a06a

Job running on windows: http://54.183.26.131:8080/job/ros2_batch_ci_windows/

It's using the windows branch of ros2/examples ros2.repos and pulling the outstanding windows branches.

tfoote, 2015-03-17:
There's one outstanding issue blocking building on windows: https://github.com/ros2/rosidl/issues/8

dirk-thomas, 2015-03-18:
I think this should be reopened because the current job is neither building nor running any tests yet.

wjwwood, 2015-03-18:
That probably should have been in the AC,  I guess it needs to do the same collecting and reporting with Jenkins like the Linux batch job.

wjwwood, 2015-03-18:
@tfoote Make sure to update `setuptools`, otherwise the scripts for things like `ament_copyright` won't get installed.

tfoote, 2015-03-18:
Created #24 to capture this

closed 2015-03-18 by tfoote

wjwwood, 2015-03-18:
Actually, they're still not getting installed. I'll look at it with @dirk-thomas tomorrow.

dirk-thomas, 2015-03-31:
The windows slave lacks `flake8`: search for `ImportError` in http://54.183.26.131:8080/view/ros2/job/ros2_batch_ci_windows/84/consoleFull

wjwwood, 2015-03-31:
I installed flake8, added a note to the internal wiki, and updated the configuration so that it should (hopefully) ignore the return code of the last two commands and mark it unstable.