ros2/ros2 #9: Get Windows compiling working

wjwwood, 2015-03-05:
Acceptance Criteria:

- [ ] Get `ament` tools running.
- [ ] Get a simple CMake project building and installing with ament tools.
- [ ] Get all packages (listed in `ros2.repos`) building and installing.
- [ ] Create document to describe how to use Windows to build ros2.

wjwwood, 2015-03-07:
Ok, so here's an update:

I've got the tools running and I can build up to the `builtin_msgs` package! The `builtin_msgs` package fails on a C++ compiler error which I think might even be fixed on master, but I haven't had time to update my sources lately. What that means to me is that most if not all of our build infrastructure is working, which is pretty exciting.

I've got some pretty straight forward instructions going on the wiki:

https://github.com/ros2/examples/wiki/Windows-Development-Setup

I have quite a few patches so far, I'll start opening pull requests next.

wjwwood, 2015-03-07:
Ok, after pushing a bit more on it, it turns out linking is not working. Exporting shared libraries on Windows turns out to be rather complicated, or at least unfamiliar to me.

I'm giving up for now, but hopefully I can wrap this up in the first part of next week.

wjwwood, 2015-03-10:
Another update. I have all but one of the `*setup.bat` files converted to `.bat`. And of the ones I've converted, only the prefix level `local_setup.bat.in` has a todo left.

I have not been able to resolve the exporting of shared libraries across packages yet, though I didn't spend much time on that today. I think that if we can get over that hurdle then we'll be close to having the tool chain fully working.

wjwwood, 2015-03-12:
Another status update!

There's good new and bad news, I'll start with good. It works! Here's a screen capture:

https://vimeo.com/121977177

So then the bad news. To get this to work I had to use the newest VS, VS 14.0 2015 which is still a technical preview, but should released in this quarter. Basically VS 12.0 2013 is just an incomplete implementation of C++11, and it is missing several critical features we need. I've updated the instructions to point out we need the newer version. Even though our opensplice package that we have for Windows is for VS2013, it seems to work alrightwith VS2015 having only needed to patch one minor thing in a header.

----

Some things that are yet to be done:

- Fix Service examples, they aren't compiling right now, but I think a similar fix I made for Topics will work.
- I need to shape up the outstanding pull requests and summarize what is done and what isn't.
- Test Connext (I need the special Services library from them).
- Push one patch upstream to OpenSplice.
- Push both RTI and PrismTech to get VS 2015 binaries out asap.

But I'll start that tomorrow...

wjwwood, 2015-03-14:
Ok, I've got all examples in the `userland` package compiling and at least the Connext ones are running. I am having an issue with the xtypes and opensplice versions where they crash for some reason. I think that at least the opensplice issue is not isolated to Windows, @esteve mentioned something about crashing on registration of String, which is the same issue I am having now. Clearly it worked before, since I made a video of it.

I've also testing Connext and it's building and running in most cases.

Still outstanding issues:

- Push one of my patches upstream to OpenSplice (it's listed in the Windows instructions for now).
- Complete some TODO's for this pull request: https://github.com/ament/ament_package/pull/6#issue-60469101

However, at this point it is in a working state, so I think we should start merging these prs and make new issues for outstanding issues.

So @esteve @tfoote @dirk-thomas please review the associated pull requests for merge.

----

On a side note I had to make a change to rclcpp in order to work around an issue with the VS 2015 compiler, see: https://connect.microsoft.com/VisualStudio/feedback/details/1179590

esteve, 2015-03-16:
Unrelated to this ticket, but more info about the issue that @wjwwood mentions in his comment. I found out that the OpenSplice examples are linked against Connext too:

```
esteve@osrf-esteve:~/Projects/ros2/ros2_ws$ ldd ./install/bin/listener__rmw_opensplice_cpp 
        linux-vdso.so.1 =>  (0x00007fff00255000)
        librmw_opensplice_cpp.so => /home/esteve/Projects/ros2/ros2_ws/install/lib/librmw_opensplice_cpp.so (0x00007f5a1a6fb000)
        libsimple_msgs__dds_connext_cpp.so => /home/esteve/Projects/ros2/ros2_ws/install/lib/libsimple_msgs__dds_connext_cpp.so (0x00007f5a1a279000)
        libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f5a1a02e000)
        libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f5a19d2a000)
        libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f5a19b13000)
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f5a1974e000)
        libdcpssacpp.so => /usr/lib/libdcpssacpp.so (0x00007f5a19430000)
        /lib64/ld-linux-x86-64.so.2 (0x00007f5a1a905000)
        libbuiltin_msgs__dds_connext_cpp.so => /home/esteve/Projects/ros2/ros2_ws/install/lib/libbuiltin_msgs__dds_connext_cpp.so (0x00007f5a191fa000)
        libnddscpp.so => /home/esteve/Projects/rti/usbstick/ubuntu/ndds.5.1.0/lib/x64Linux3.xgcc4.6.3/libnddscpp.so (0x00007f5a18eba000)
        libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f5a18bb3000)
        libdcpsgapi.so => /usr/lib/libdcpsgapi.so (0x00007f5a1894b000)
        libddsdatabase.so => /usr/lib/libddsdatabase.so (0x00007f5a1870f000)
        libddsos.so => /usr/lib/libddsos.so (0x00007f5a184ec000)
        libddskernel.so => /usr/lib/libddskernel.so (0x00007f5a18259000)
        libddsuser.so => /usr/lib/libddsuser.so (0x00007f5a1801b000)
        librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007f5a17e12000)
        libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f5a17c0e000)
        libnddsc.so => /home/esteve/Projects/rti/usbstick/ubuntu/ndds.5.1.0/lib/x64Linux3.xgcc4.6.3/libnddsc.so (0x00007f5a175ce000)
        libnddscore.so => /home/esteve/Projects/rti/usbstick/ubuntu/ndds.5.1.0/lib/x64Linux3.xgcc4.6.3/libnddscore.so (0x00007f5a16ecd000)
        libddsutil.so => /usr/lib/libddsutil.so (0x00007f5a16cc3000)
        libddsserialization.so => /usr/lib/libddsserialization.so (0x00007f5a16a99000)
        libddsconf.so => /usr/lib/libddsconf.so (0x00007f5a16895000)
        libddsconfparser.so => /usr/lib/libddsconfparser.so (0x00007f5a16688000)
```

wjwwood, 2015-03-21:
Ok, Windows is not building, but I need to fix some structural problems. Currently linux is working with these branches so I'm going to merge and open new pull requests to fix Windows.

closed 2015-03-21 by wjwwood