ros2/ros2 #12: Configure and Enable the cpplint QA tool

wjwwood, 2015-03-10:
We have a tools for checking our code with `clang_format` but it is not configured to our liking yet.

Acceptance Criteria:

- [ ] Decide on a configuration with the team
- [ ] Reenable the linter
- [ ] Create tasks for fixing failing packages

dirk-thomas, 2015-10-15:
Decided on configuration in ament/ament_lint#31

dirk-thomas, 2015-10-15:
Updated repos:
- [x] ament/ament_index#6
- [x] ros2/rosidl#42
- [x] ros2/rosidl_dds#27

jacquelinekay, 2015-10-20:
I will take rttest

jacquelinekay, 2015-10-20:
@wjwwood will take rclcpp (and other rcl libraries?)

dirk-thomas, 2015-10-26:
Related ticket to either ignore `runtime/int` errors or update the code to comply with `cpplint`: ament/ament_lint#33

dirk-thomas, 2015-11-04:
The latest job with cpplint enabled: http://ci.ros2.org/job/ros2_batch_ci_linux/547/ I have update the todo list in the first comment to reflect the repos which need fixing.

closed 2015-11-06 by dirk-thomas