ros2/ros2 #20: Use white space in path for batch CI jobs

wjwwood, 2015-03-17:


tfoote, 2015-03-31:
Blocked on ament/ament_tools#27

tfoote, 2015-04-01:
I've submitted several patches. However I've run into a problem in that idlpp from opensplice is not whitespace safe. When invoked with quoted strings. (using  ros2/rmw_opensplice#22  and the other patches)

```
/usr/bin/idlpp -I "/home/tfoote/work/r2/ros2 ws/build/simple_msgs/rosidl_generator_dds_idl" -S -l cpp -o dds-types -d "/home/tfoote/work/r2/ros2 ws/build/simple_msgs/rosidl_typesupport_opensplice_cpp/simple_msgs/dds_opensplice" "/home/tfoote/work/r2/ros2 ws/build/simple_msgs/rosidl_generator_dds_idl/simple_msgs/dds_opensplice/Time_.idl"
```
It internally invokes cppgen with unquoted paths.
```
cppgen -I"/home/tfoote/work/r2/ros2 ws/build/simple_msgs/rosidl_generator_dds_idl" -I/home/tfoote/work/r2/ros2 ws/build/simple_msgs/rosidl_generator_dds_idl/simple_msgs/dds_opensplice -I/usr/etc/idl -I/usr/etc/opensplice/idlpp -ignore_interfaces -output=/home/tfoote/work/r2/ros2 ws/build/simple_msgs/rosidl_typesupport_opensplice_cpp/simple_msgs/dds_opensplice /home/tfoote/work/r2/ros2 ws/build/simple_msgs/rosidl_generator_dds_idl/simple_msgs/dds_opensplice/Time_.idl
```

This is something out of our control

dirk-thomas, 2015-04-01:
@tfoote Can you please fill this upstream so that it might be addressed in future versions of OpenSplice.

tfoote, 2015-04-01:
filed upstream, see link above

tfoote, 2015-04-07:
Blocked waiting for upstream. tracking a patch at: https://github.com/ros2/ros2/issues/35

tfoote, 2015-04-14:
Close in favor of #36

closed 2015-04-14 by tfoote