ros2/ros2 #21: Fix all test failures on Linux CI Batch Job

wjwwood, 2015-03-17:
AC:

- [ ]  All tests should pass without failure for the Linux CI Batch Job

dirk-thomas, 2015-04-01:
https://github.com/ros2/rosidl_dds/commit/64a1f36785ee31c3584b6a0adf3d1ca069f92c66
https://github.com/ros2/rmw_connext/commit/57cf7f523d6788980bea118ba1828b1a996fa70d
https://github.com/ros2/rmw_opensplice/commit/cb75a3d995e78f44df0087698eab5b96defb40a9

wjwwood, 2015-04-01:
Just FYI, there is another CMake linter I found the other day: https://github.com/polysquare/polysquare-cmake-linter

dirk-thomas, 2015-04-07:
@esteve @tfoote @wjwwood Please take a close look to the reference PRs which apply the uncrustify formatting rules. Even if it is a lot of white spaces changes we should make sure that the formatting is exactly what we want - and if necessary adjust it before merging them.

dirk-thomas, 2015-04-07:
I have filled the following tickets upstream:

* https://github.com/bengardner/uncrustify/issues/362
* https://github.com/bengardner/uncrustify/issues/363

dirk-thomas, 2015-04-07:
* missing Connext

dirk-thomas, 2015-04-08:
I have updated the list in the first comment after looking through the currently failing tests.

wjwwood, 2015-04-15:
I finally fixed the communication problem within docker, I had to do `docker run` with the `--privileged` option and I had to add an `ifconfig eth0 multicast` as part of the run (not possible in the build part of the docker image and it requires `--privileged`). After that the userland talker/listener test passes.

Now the only failing tests on this Linux build are uncrustify related:

http://54.183.26.131:8080/job/ros2_batch_ci_linux/13/console

Though it still doesn't find Connext, so there maybe more failures lurking in the connext packages still.

wjwwood, 2015-04-15:
This build is green: http://54.183.26.131:8080/job/ros2_batch_ci_linux/

wjwwood, 2015-04-15:
:fireworks: Yay! :fireworks:

I'll close this now, deferring to ros2/ros2#33 and ros2/ros2#34 for OS X and Windows respectively.

closed 2015-04-15 by wjwwood

dirk-thomas, 2015-04-16:
The current linux batch job does not use Connext.

dirk-thomas, 2015-04-17:
The current linux batch job (http://54.183.26.131:8080/view/ros2/job/ros2_batch_ci_linux/27/) fails the test_publish_subscribe with OpenSplice. It seems that both endpoints can't communicate again?

wjwwood, 2015-04-17:
@dirk-thomas maybe it's something else this time. Last time it said in the logs that multicast was not available in the ospl log, but now it doesn't:

http://54.183.26.131:8080/view/ros2/job/ros2_batch_ci_linux/ws/workspace/src/ros2/examples/userland/ospl-info.log

(Btw the tests drop the ospl log in the source space)

The next thing to do would be to log in to the docker instance locally and run the talker and listener in different folders since they some times clobber each other writing to the same ospl log file.