ros2/ros2 #22: Package up RTI's distribution with the extra Messaging library

wjwwood, 2015-03-17:
AC:

- [ ] Able to `apt-get install` RTI with all of the extensions we need to build our stuff.

esteve, 2015-04-01:
@dirk-thomas @tfoote @wjwwood I packaged the messaging library at https://github.com/osrf/rticonnextdds-messaging, let me know if it works for you. It's not `apt-get`'able because it's private and I didn't think we needed the infrastructure, so where should we put the generated .debs?

wjwwood, 2015-04-01:
As a download on a release tag, like we've attached things for OpenSplice: https://github.com/osrf/opensplice/releases/tag/6.4.0-0

esteve, 2015-04-01:
@wjwwood Done: https://github.com/osrf/rticonnextdds-messaging/releases/tag/ubuntu%2Ftrusty%2F5.1.0

esteve, 2015-04-03:
New release:

https://github.com/osrf/rticonnextdds-messaging/releases/tag/ubuntu/trusty/5.1.0-2

closed 2015-04-06 by esteve

tfoote, 2015-04-06:
Note you also need these packages to build upon: https://github.com/osrf/rticonnextdds-src/releases/tag/5.1.0-2