ros2/ros2 #23: Make batch job flexible to look for same branch name on other repos

tfoote, 2015-03-17:
AC: 

- [ ] upgrade batch job to detect PR branch and then check out other repos with the same branch name it it exists else fallback to default branch.

tfoote, 2018-03-08:
@nuclearsandwich This is probably deprecated by the goals of getting full CI running for ROS2?

nuclearsandwich, 2018-03-08:
> This is probably deprecated by the goals of getting full CI running for ROS2?

As I understand it we can handle jobs like this now by triggering a CI launcher build with a custom branch name. The additional feature of this request would be the automatic triggering of CI on new pull requests.

For ROS 2 my understanding is that we want devel jobs to be able to (to always?) pull in sibling branches, which ROS 1 devel jobs don't currently do.

With current infrastructure priorities for ROS 2 such jobs are second only to overall CI time reduction but both are blocked on the build tooling discussion.