ros2/ros2 #24: Add testing to Batch CI for Windows

tfoote, 2015-03-18:
AC: 

- [ ] Tests are run
- [ ] Test results are summarized in jenkins 

----

Follow up to ros/ros2#6

tfoote, 2015-03-21:
It's there and turning over and trying to test. I'd failing to build the tests at the moment. re: ament/ament_tools#23

It's configured to collect the tests, but they aren't being generated yet.

wjwwood, 2015-03-31:
The windows job is turning over now, I propose we close this one.

@dirk-thomas @esteve @tfoote

esteve, 2015-03-31:
@wjwwood Agreed.

dirk-thomas, 2015-03-31:
The current job does not have a single successfully running test. Therefore I don't think this should be closed yet.

wjwwood, 2015-03-31:
> The current job does not have a single successfully running test.

The `osrf_common` tests are running and being captured.

> Therefore I don't think this should be closed yet.

I would have preferred to open a new issue rather than continuously extend the criteria of this issue, but since we've already done it, we can leave it I guess.

dirk-thomas, 2015-03-31:
Done.

closed 2015-03-31 by dirk-thomas