ros2/ros2 #25: Move implementation of OS X CI Batch job to a repository

wjwwood, 2015-03-18:
Currently the tasks are contained completely in the Jenkins job configuration.

AC:

- [ ] The steps to build the OS X batch CI moved to a repository.

----

Should we create a repository with the scripts for the batch CI jobs? Currently there is a repo on @tfoote's account: https://github.com/tfoote/ros2_batch_ci which contains the Docker stuff and Linux CI batch logic.

dirk-thomas, 2015-03-24:
Put the content as well as the ros2_batch_ci content on a branch in the ros2/ros2 repo.

dirk-thomas, 2015-03-24:
Beside refactoring the logic into a script (rather then multiple commands in the job) it would also be beneficial to automate the whole job deployment.

Using `ros_buildfarm` it should be only a matter of a couple of lines of Python code (invoking https://github.com/ros-infrastructure/ros_buildfarm/blob/baee67802c4da253f7935b0be0f0d6ed35d4e1c4/ros_buildfarm/jenkins.py#L103 and https://github.com/ros-infrastructure/ros_buildfarm/blob/baee67802c4da253f7935b0be0f0d6ed35d4e1c4/ros_buildfarm/jenkins.py#L168). You can easily get the job configuration xml by going to the job configure page in Jenkins and replace `configure` with `config.xml`. That way it also include the job configuration details which are not covered by the script.

wjwwood, 2015-03-28:
I've moved (and improved) the implementation of the batch script to the ros2/ros2 repository and updated the job:

- https://github.com/ros2/ros2/tree/ci_scripts
- https://github.com/ros2/ros2/blob/ci_scripts/osx_batch/osx_batch_build.py#L67-L128

Here is the configuration of the job as it is:

```xml

<?xml version='1.0' encoding='UTF-8'?>
<project>
  <actions/>
  <description></description>
  <keepDependencies>false</keepDependencies>
  <properties>
    <com.coravy.hudson.plugins.github.GithubProjectProperty plugin="github@1.10">
      <projectUrl>https://github.com/ros2/ros2/</projectUrl>
    </com.coravy.hudson.plugins.github.GithubProjectProperty>
    <jenkins.advancedqueue.AdvancedQueueSorterJobProperty plugin="PrioritySorter@2.9">
      <useJobPriority>false</useJobPriority>
      <priority>-1</priority>
    </jenkins.advancedqueue.AdvancedQueueSorterJobProperty>
    <org.jenkinsci.plugins.requeuejob.RequeueJobProperty plugin="jobrequeue@1.0">
      <requeueJob>false</requeueJob>
    </org.jenkinsci.plugins.requeuejob.RequeueJobProperty>
  </properties>
  <scm class="hudson.plugins.git.GitSCM" plugin="git@2.3.4">
    <configVersion>2</configVersion>
    <userRemoteConfigs>
      <hudson.plugins.git.UserRemoteConfig>
        <url>https://github.com/ros2/ros2.git</url>
      </hudson.plugins.git.UserRemoteConfig>
    </userRemoteConfigs>
    <branches>
      <hudson.plugins.git.BranchSpec>
        <name>*/ci_scripts</name>
      </hudson.plugins.git.BranchSpec>
    </branches>
    <doGenerateSubmoduleConfigurations>false</doGenerateSubmoduleConfigurations>
    <browser class="hudson.plugins.git.browser.GithubWeb">
      <url></url>
    </browser>
    <submoduleCfg class="list"/>
    <extensions/>
  </scm>
  <assignedNode>osx_slave</assignedNode>
  <canRoam>false</canRoam>
  <disabled>false</disabled>
  <blockBuildWhenDownstreamBuilding>false</blockBuildWhenDownstreamBuilding>
  <blockBuildWhenUpstreamBuilding>false</blockBuildWhenUpstreamBuilding>
  <triggers/>
  <concurrentBuild>false</concurrentBuild>
  <builders>
    <hudson.tasks.Shell>
      <command>/usr/local/bin/python3 -u osx_batch/osx_batch_build.py</command>
    </hudson.tasks.Shell>
  </builders>
  <publishers>
    <xunit plugin="xunit@1.93">
      <types>
        <GoogleTestType>
          <pattern>workspace/build/*/test_results/**/*.xml</pattern>
          <skipNoTestFiles>false</skipNoTestFiles>
          <failIfNotNew>true</failIfNotNew>
          <deleteOutputFiles>true</deleteOutputFiles>
          <stopProcessingIfError>true</stopProcessingIfError>
        </GoogleTestType>
      </types>
      <thresholds>
        <org.jenkinsci.plugins.xunit.threshold.FailedThreshold>
          <unstableThreshold>0</unstableThreshold>
          <unstableNewThreshold></unstableNewThreshold>
          <failureThreshold></failureThreshold>
          <failureNewThreshold></failureNewThreshold>
        </org.jenkinsci.plugins.xunit.threshold.FailedThreshold>
        <org.jenkinsci.plugins.xunit.threshold.SkippedThreshold>
          <unstableThreshold>0</unstableThreshold>
          <unstableNewThreshold></unstableNewThreshold>
          <failureThreshold></failureThreshold>
          <failureNewThreshold></failureNewThreshold>
        </org.jenkinsci.plugins.xunit.threshold.SkippedThreshold>
      </thresholds>
      <thresholdMode>1</thresholdMode>
      <extraConfiguration>
        <testTimeMargin>3000</testTimeMargin>
      </extraConfiguration>
    </xunit>
  </publishers>
  <buildWrappers>
    <hudson.plugins.ansicolor.AnsiColorBuildWrapper plugin="ansicolor@0.4.1">
      <colorMapName>xterm</colorMapName>
    </hudson.plugins.ansicolor.AnsiColorBuildWrapper>
  </buildWrappers>
</project>
```

I'm posting it here for posterity, and I'm going to punt the automatic configuration of the job using the `ros_buildfarm` API to another issue. I want to do this because I think it's not clear what the mapping from github repositories and jenkins jobs will be yet and that will affect the design of the "deploy" script.

wjwwood, 2015-03-31:
I'll go ahead and create the `ros_buildfarm` script to create this job, since something more complicated will probably take too long.

But I've run into some problems.

@dirk-thomas The `ros_buildfarm.templates.expand_template` function does not allow you to pass in an external template file nor a string for the template. There also does not appear to be a similar function in the module to achieve this. Is there a way to provide an external template?

Also, I was basing the template for this job on the above output, but it seems like the other templates make use of a lot of `SNIPPETS`, should I start with one of those instead?

dirk-thomas, 2015-03-31:
@wjwwood The ros_buildfarm templates have never been used outside of the repository. Probably the best way would be to allow extending `template_basepath` with a search path like list of base paths?

Each job template can choose to reuse snippets if it makes sense and avoid repetition but can also be completely self contained. Since the current jobs are only temporary solutions I don't have a preference - dumping the complete job xml is fine with me to save effort.

wjwwood, 2015-03-31:
Ok, I just created a command which will push the raw configuration posted above. I'm testing it here: http://54.183.26.131:8080/job/ros2_batch_ci_osx_testing/1/console

I think this is ready to close.

wjwwood, 2015-03-31:
Bump, @dirk-thomas @esteve @tfoote.

dirk-thomas, 2015-03-31:
The job lgtm. The script uses some relative path (`../../ros_buildfarm`) as a fallback which looks weird. Should that be removed?

wjwwood, 2015-03-31:
I think it's convenient to be able to just clone them as peers, but I suppose it could be surprising if you aren't expecting for it to import from there. I'll remove it.

closed 2015-03-31 by wjwwood