ros2/ros2 #26: Fix linking on Windows

wjwwood, 2015-03-21:
After discussing with @esteve on the whiteboard, I'm to clean up (one more time) the symbols being used for "getting type support handles". The current setup works on Linux and could on Windows, but is pretty confusing and inconsistent.

I'll follow up with the summary of our whiteboard work after this.

AC:

- [ ] Compiling and Linking work on Linux and Windows.
- [ ] Implementation specific executables do not cross talk (the linker is still able to choose the correct library).

wjwwood, 2015-03-25:
Ok, I've refactored the way we define the `get_type_support_handle` function across different implementations of rmw.

With these pr's merged, I can compile and run both the opensplice and connext executables on Windows, Linux, and OS X. I double checked that we don't have the linking cross talk either.

One outstanding problem is that I haven't updated the `get_service_type_support_handle` functions to follow the same pattern, but they seem to be working as is and so I didn't mess with that as they are also not implemented in all cases (opensplice yet and connext xtypes). Also I was unable to compile the `.srv` files and the Service examples on OS X due to some problem with the Connext libraries on OS X. I'll have to open a ticket about that.

But for now, I think this issue and the associated pull requests are ready for review.

@esteve @dirk-thomas @tfoote

(also I might be late tomorrow, I just sort of looked up and it was 4am...)

wjwwood, 2015-03-25:
Ok, I'm refactoring these branches a bit after discussing with @dirk-thomas and @esteve.

This is the new plan:

- `rosidl_generator_c` provides:
 - `rosidl_generator_c/message_type_support.h`: which defines `rosidl_message_type_support_t`
 - `rosidl_generator_c/service_type_support.h`: which defines `rosidl_service_type_support_t`
- `rosidl_generator_cpp` provides:
 - `rosidl_generator_cpp/message_type_support_decl.hpp`: which defines the function `const rosidl_message_type_support_t * rosidl_generator_cpp::get_message_type_support_handle()`
 - `rosidl_generator_cpp/service_type_support_decl.hpp`: which defines the function `const rosidl_service_type_support_t * rosidl_generator_cpp::get_service_type_support_handle()`
- Each implementation's `rosidl_typesupport_<impl>_cpp` package provides (in a normally hidden include dir):
 - `rosidl_generator_cpp/message_type_support.hpp`: which includes `rosidl_generator_cpp/message_type_support_decl.hpp` and implements the function it defines with an implementation specific version named `rosidl_typesupport_<impl>_cpp::get_message_type_support_handle_<impl>()`.
 - `rosidl_generator_cpp/service_type_support.hpp`: which includes `rosidl_generator_cpp/service_type_support_decl.hpp` and implements the function it defines with an implementation specific version named `rosidl_typesupport_<impl>_cpp::get_service_type_support_handle_<impl>()`.
- Finally, `rcl`/`rclcpp` include `<rosidl_generator_cpp/{message, service}_type_support.hpp>` and use the function `rosidl_generator_cpp::get_{message, service}_type_support_handle()`

The dependencies will be update to reflect this:

- `rosidl_generator_cpp` build and build_export depend on `rosidl_generator_c`
- `rosidl_typesupport_*` packages build and build_export depend on `rosidl_generator_cpp` and `rosidl_generator_c`
- `rcl`/`rclcpp` build and build_export depend on `rosidl_default_generators` and `rosidl_generator_c` and `rosidl_generator_cpp`.

If you guys could look this plan over and give me feedback before I implement it that'd be great.

dirk-thomas, 2015-03-25:
LGTM

Additionally `rosidl_generator_cpp` can provide:
* `rosidl_generator_cpp/{message, service}_type_support.hpp`: which contains an `#error` directive pointing out that no implementation is on the path

esteve, 2015-03-25:
LGTM too, plus @dirk-thomas comment

wjwwood, 2015-03-26:
Ok the new plan has been implemented on each of the attached branches as new commits. I tried to keep the non-plan related changes to separate commits, so you might want to review by commit.

I'm testing on Windows now, I'll test on Linux too after that.

dirk-thomas, 2015-03-26:
The new commits on all 4 PRs look good to me. Once it works on all platforms `rosidl_generator_cpp` still needs the fallback header with the `#error`.

wjwwood, 2015-03-26:
@dirk-thomas I forgot about that, I should test it now.

wjwwood, 2015-03-26:
Looks like it works fine, also testing is complete on Windows and Linux, I'm going to go ahead with merging, feel free to do post merge reviews, we can also follow up.

wjwwood, 2015-03-26:
@dirk-thomas I spoke too soon on the headers with `#error` warnings in them. It seems they will always get installed to `<prefix>/include` when inevitably ends up ahead of the other includes. I'll have to remove those from master.

wjwwood, 2015-03-26:
I removed them in 24ed8cc, we can think about another way to have a trip wire header like that tomorrow.

dirk-thomas, 2015-07-24:
It is currently failing again since we now have more than two message packages which depend on each other. The problem is that the generated code for Connext as well as OpenSplice does not set dllexport attribute correctly.

dirk-thomas, 2015-07-24:
With the newly referenced PRs the CI job finishes linking: http://ci.ros2.org/job/ros2_batch_ci_windows/45/

closed 2015-07-24 by dirk-thomas