ros2/ros2 #30: revisit type support identifiers

dirk-thomas, 2015-03-30:
After the refactoring to support C as well as Windows the type support identifiers are not working as they were before.

E.g. for Connext XTypes a string comparison is necesssary (ros2/rmw_connext#2). For other type supports the check is not being performed anymore.

wjwwood, 2015-03-30:
Yeah, this is my fault, I wasn't aware that the identifier check was being done in this manner (not using `strcmp`). Hopefully this kind of thing will happen less often once we get some CI. Ideally when we revisit this we'll have some automated test which ensures it is still working.

dirk-thomas, 2015-04-13:
The two PRs address the Connext XTypes only. I still have to look for a similar solution for the other type supports.

dirk-thomas, 2015-04-14:
Two more PRs to readd typesupport checking to static connext and opensplice.

Ready to be reviewed.

wjwwood, 2015-04-15:
Are there any more outstanding PR's that need to be made for this? If not I think it's good to move to 'done'.

closed 2015-04-15 by dirk-thomas