ros2/ros2 #31: Link error for Connext services on OS 10.10

wjwwood, 2015-03-31:
On OS X 10.10 I get this linkedit error:

```
Undefined symbols for architecture x86_64:
  "connext::check_retcode(DDS_ReturnCode_t, char const*, RTILogMessage const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, bool)", referenced from:
      connext::details::SampleBase<userland_msgs::dds_::AddTwoIntsResponse_, DDS_WriteParams_t>::initialize() in AddTwoInts_ServiceTypeSupport.cpp.o
      connext::details::SampleBase<userland_msgs::dds_::AddTwoIntsResponse_, DDS_WriteParams_t>::copy_from(userland_msgs::dds_::AddTwoIntsResponse_ const&) in AddTwoInts_ServiceTypeSupport.cpp.o
      connext::details::SampleBase<userland_msgs::dds_::AddTwoIntsResponse_, DDS_SampleInfo>::initialize() in AddTwoInts_ServiceTypeSupport.cpp.o
      connext::details::SampleBase<userland_msgs::dds_::AddTwoIntsResponse_, DDS_SampleInfo>::copy_from(userland_msgs::dds_::AddTwoIntsResponse_ const&) in AddTwoInts_ServiceTypeSupport.cpp.o
      void connext::details::copy_sample<userland_msgs::dds_::AddTwoIntsResponse_, DDS_SampleInfo>(connext::Sample<userland_msgs::dds_::AddTwoIntsResponse_>&, userland_msgs::dds_::AddTwoIntsResponse_ const&, DDS_SampleInfo const&) in AddTwoInts_ServiceTypeSupport.cpp.o
      connext::LoanedSamples<userland_msgs::dds_::AddTwoIntsResponse_>::move_construct_from_loans(userland_msgs::dds_::AddTwoIntsResponse_DataReader*, userland_msgs::dds_::AddTwoIntsResponse_Seq&, DDS_SampleInfoSeq&) in AddTwoInts_ServiceTypeSupport.cpp.o
      connext::details::SampleBase<userland_msgs::dds_::AddTwoIntsRequest_, DDS_SampleInfo>::initialize() in AddTwoInts_ServiceTypeSupport.cpp.o
      ...
  "connext::RequesterParams::service_name(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&)", referenced from:
      userland_msgs::service_type_support::create_requester__AddTwoInts(void*, char const*, void**) in AddTwoInts_ServiceTypeSupport.cpp.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
```

My current theory is that the shared libraries provided by RTI was not compiled with the same version of the std library as I have here on 10.10. I'd like to reach out to RTI on this one.

Until this gets resolved, the OS X CI job will continue to fail.

wjwwood, 2015-04-01:
I looked at this with @gerkey and poked around on the internet. It seems there is good evidence supporting the idea that my version of `std::string`, which is defined in `libc++`, generates a different symbol from the `std::string` with the GNU std library called `libstdc++`.

I cannot work around this by choosing to use `libstdc++` because on OS X it does not support C++11, which we are using. Therefore I will need a new version of the RTI libraries which are built against `libc++` rather than `libstdc++`. Here are some supporting materials for this conclusion:

- http://stackoverflow.com/questions/8454329/why-cant-clang-with-libc-in-c0x-mode-link-this-boostprogram-options-examp/8457799#8457799
- http://stackoverflow.com/questions/12542971/using-libstdc-compiled-libraries-with-clang-stdlib-libc
- http://stackoverflow.com/questions/12920891/std-linker-error-with-apple-llvm-4-1

I'm going to contact RTI about getting a newer version of the shared libraries.

wjwwood, 2015-04-21:
TODO: re-compare test numbers (https://github.com/ros2/ros2/issues/33) after this is fixed.

dirk-thomas, 2015-04-29:
Latest job failing with this: http://54.183.26.131:8080/view/ros2/job/ros2_batch_ci_osx/95/console

    Undefined symbols for architecture x86_64:
      "connext::check_retcode(DDS_ReturnCode_t, char const*, RTILogMessage const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, bool)", referenced from:
          connext::details::type_support_adapter<DDS_DynamicData>::register_type(DDSTypeSupport*, DDSDomainParticipant*) in functions.cpp.o
          connext::LoanedSamples<DDS_DynamicData>::move_construct_from_loans(DDSDynamicDataReader*, DDS_DynamicDataSeq&, DDS_SampleInfoSeq&) in functions.cpp.o
      "connext::RequesterParams::service_name(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&)", referenced from:
          _rmw_create_client in functions.cpp.o
    ld: symbol(s) not found for architecture x86_64
    clang: error: linker command failed with exit code 1 (use -v to see invocation)

tfoote, 2015-05-05:
New package from RTI, trying to test, but it's not working out of the box.

wjwwood, 2015-05-06:
Ok, I've got it working, here is a passing job (with all the tests I would expect) with Connext turned on:

http://54.183.26.131:8080/job/ros2_batch_ci_osx/119/

closed 2015-05-06 by wjwwood