ros2/ros2 #33: Compare OSX unit test results with Linux

dirk-thomas, 2015-03-31:
Depends on ros2/ros2#21

AC:

- [ ] either equal test counts or
- [ ] known differences

dirk-thomas, 2015-04-07:
* cppcheck missing
* PEP8 test failing but not on Linux

wjwwood, 2015-04-15:
ros2/ros2#31

gerkey, 2015-04-21:
Can be closed with a lower test count if we can show that the missing tests are due to the lack of Connext, because of ros2/ros2#31.

wjwwood, 2015-04-21:
Linux (without Connext) currently has 408 tests:

http://54.183.26.131:8080/job/ros2_batch_ci_linux/50/testReport/

OS X has 401 tests:

http://54.183.26.131:8080/job/ros2_batch_ci_osx/93/testReport/

I'm looking at the differences.

wjwwood, 2015-04-21:
Actually after running an up-to-date build of OS X it also has 408 now:

http://54.183.26.131:8080/job/ros2_batch_ci_osx/94/testReport/

closed 2015-04-21 by wjwwood