ros2/ros2 #34: Compare Windows unit test results with Linux

wjwwood, 2015-03-31:
Depends on ros2/ros2#21

AC:

- [ ] either equal test count or
- [ ] different test count with rational differences

dirk-thomas, 2015-04-07:
* clang-format missing
* uncrustify missing
* failing PEP8 test but not on Linux?

wjwwood, 2015-04-15:
ros2/rmw_connext#20

dirk-thomas, 2015-04-16:
I have updated the uncrustify wrapper package to work on windows (supposingly): https://github.com/ament/uncrustify/commit/96f42bcda8e60d705a028a0f73a7d3b48b327a2f

@wjwwood Currently Visual Studio on the Windows slave does not work anymore. Once this is fixed please trigger another build to check if it makes the uncrustify tests pass.

dirk-thomas, 2015-04-17:
All uncrustify tests: passing

wjwwood, 2015-06-16:
Linux and Windows now have the same number of tests (530):

- http://54.183.26.131:8080/job/ros2_batch_ci_windows/288/testReport/
- http://54.183.26.131:8080/job/ros2_batch_ci_linux/145/testReport/

There are some skipped tests and failing tests on Windows, but we have the same number at least. I scan over the tests and it appears that all of the failing tests on Windows are present on Linux and that there is not any linux only or windows only tests. Closing this, please comment if you see a problem.

closed 2015-06-16 by wjwwood