ros2/ros2 #36: Unify Jenkins batch jobs

dirk-thomas, 2015-04-07:


dirk-thomas, 2015-04-08:
We should consider to generate multiple jobs per platform. E.g. on Linux we should have one job using the official Connext binary package and another one using our custom Debian package.

wjwwood, 2015-04-09:
Ok, I've parameterized a version of the Windows job, please take a look and if there are no suggestions, I'll move forward making the OS X and Linux jobs very similar.

wjwwood, 2015-04-09:
http://54.183.26.131:8080/job/ros2_batch_ci_windows_experimental/

dirk-thomas, 2015-04-09:
The first build with a custom branch looked good (number 16, I aborted it after seeing the correct checkout).

I then tried with spaces (http://54.183.26.131:8080/job/ros2_batch_ci_windows_experimental/17/) and that seems to fail. The PYTHONPATH seems to be wrong - likely a problem in one of the bat files.

dirk-thomas, 2015-04-09:
When switching between spaces and no spaces it also does not clean the other workspace folder from previous builds.

wjwwood, 2015-04-09:
> I then tried with spaces (http://54.183.26.131:8080/job/ros2_batch_ci_windows_experimental/17/) and that seems to fail. The PYTHONPATH seems to be wrong - likely a problem in one of the .bat files?

In this issue I'm really only concerned about whether or not the correct folders were passed to the tools, not whether or not our entire software stack support whitespace since we have outstanding known issues.

> When switching between spaces and no spaces it also does not clean the other workspace folder from previous builds.

It does, it just doesn't clear previous builds of a different configuration. If you build spaces, no spaces, then spaces again it will remove the first one with spaces. I'd rather not force the tool to remove all possible workspace folders on each run, because when using it locally I would like to switch back and forth between spaces and no spaces without clearing each other. So I've added a `rmdir /S /Q workspace "work space"` to the Jenkins job configuration.

dirk-thomas, 2015-04-10:
Since the failing job when white spaces are enabled is specific to the Windows build I would say it looks good and you can go ahead and use the same template for Linux / OS X.

wjwwood, 2015-04-14:
Ok, I think this is ready for review. I have a "launcher" job which will spawn each of the other three batch jobs, forwarding along the options:

- http://54.183.26.131:8080/job/ros2_batch_ci_launcher/

Then we have each of the jobs:

- http://54.183.26.131:8080/job/ros2_batch_ci_linux/
- http://54.183.26.131:8080/job/ros2_batch_ci_osx/
- http://54.183.26.131:8080/job/ros2_batch_ci_windows/

The jobs are all created by a Python script:

- https://github.com/ros2/ros2/blob/ci_scripts/create_jenkins_job.py

Instructions on how to deploy on a new farm are in the `README.md`:

- https://github.com/ros2/ros2/blob/ci_scripts/README.md

There are still several issues, but I think we can address those separately rather than drag this issue on:

- Linux doesn't find RTI with the current `.deb`'s, but @esteve is working on that.
- OS X doesn't build with Connext set to ON, so the launcher job could be made to let the user select 'on', 'off', or additionally 'default' for the connext option so that you can at least test Linux and Windows with Connext on without worrying about OS X breaking. Hopefully in the future such flexibility won't be needed.
- Windows, Linux, and OS X don't have the right version of uncrustify, but @dirk-thomas is working on that.

@dirk-thomas @esteve @tfoote for review.

wjwwood, 2015-04-14:
I've also merged `ci_scripts2` into `ci_scripts` and removed the extra jobs.

dirk-thomas, 2015-04-15:
The license file on the branch (https://github.com/ros2/ros2/blob/ci_scripts/LICENSE) is not identical to the template (white spaces before the first content line). Also the license file should not contain the year and osrf reference. Instead that should only be part of the source files.

dirk-thomas, 2015-04-15:
The script `create_jenkins_job.py` contains a lot of xml for the job configuration. Should this be extracted into "snippet" files for easier reading / editing? Similar as for the ros_buildfarm repo.

wjwwood, 2015-04-15:
LICENSE file has been updated and I moved the job xml templates out of the python script.

esteve, 2015-04-15:
The Linux slave now finds Connext properly: http://54.183.26.131:8080/view/ros2/job/ros2_batch_ci_linux/15/console

wjwwood, 2015-04-15:
@esteve can you make a pull request to merge your version of `ci_scripts`?

esteve, 2015-04-15:
@wjwwood https://github.com/ros2/ros2/pull/39

closed 2015-04-15 by wjwwood