ros2/ros2 #38: Enhance CI launcher job to pass "default" parameter

wjwwood, 2015-04-14:
Currently you can set the same parameters on the launcher job as on the OS specific jobs, but there is not a way to pass only some options to the downstream OS specific jobs. For example, since the launcher job has the USE_CONNEXT option, the downstream jobs have to all use the same setter, so you couldn't have it on for Linux and Windows but off for OS X.

wjwwood, 2015-04-14:
I'm not sure what the right technical approach is for this, I guess you could duplicate all of the options on the launcher job, one set for each OS, but it's not clear to me how to pass those down the other jobs as the right parameters without also naming the parameters in each job with an OS specific name.

dirk-thomas, 2015-08-21:
Do we still have a need for this? Currently all platforms use the same default arguments and the user can always trigger the individual jobs separately.

wjwwood, 2015-08-21:
Should be fine to close.

closed 2015-08-21 by wjwwood