ros2/ros2 #40: Audit build caching

wjwwood, 2015-04-15:
It seems that repeat builds of the ROS 2 stack with `ament build` will result in rebuilding or even regenerating code in parts of the system.

See: https://github.com/ament/ament_tools/pull/32#issuecomment-93581112

dirk-thomas, 2015-04-20:
The first item to address is that templated message files are always overwritten. The above four PRs change that so that the generated files are only overwritten when their content actually changes. Otherwise they are not touched.

dirk-thomas, 2015-04-21:
I can't find any unnecessary operations in the log of a repeated build anymore.

@esteve @tfoote @wjwwood Please run a repeated build (with the above PRs checked out / merged) and review the build log. Then either confirm that this can be closed or comment with specific parts from the build log which should be skipped.

tfoote, 2015-04-21:
These individually looks reasonable. However there's a lot of repeated code snippets for the write file if changed which would be much better to consolidate into a function. (11 instances)

dirk-thomas, 2015-04-21:
@tfoote I agree that refactoring the functionality would be desired but consider that for a separate PR.

tfoote, 2015-04-21:
separate is fine by me ticketed: #44 to not forget.

dirk-thomas, 2015-04-22:
Closing for now.

closed 2015-04-22 by dirk-thomas