ros2/ros2 #42: Write unit test using primitive types

gerkey, 2015-04-21:


dirk-thomas, 2015-04-28:
* Create binaries for:
  * [x] publish / subscribe a message with a primitive type
  * [ ] publish / subscribe a message with fields for all primitive types
  * [ ] publish / subscribe a message with fields for all primitive types
  * [ ] publish / subscribe a message with fields for static arrays of all primitive types
  * [ ] publish / subscribe a message with fields for dynamic arrays of all primitive types
  * [ ] publish / subscribe a message with fields for all builtin types, 
  * [ ] publish / subscribe a message with nested types
  * [x] request / reply a message with a primitive type
* Create tests for:
  * [x] publish / subscribe for each rmw implementation
  * [x] publish / subscribe for between each combination of rmw implementations
  * [x] request / reply for each rmw implementation
  * [x] request / reply for between each combination of rmw implementations

dirk-thomas, 2015-05-06:
Request / reply tests currently blocked by ros2/rclcpp#29.

dirk-thomas, 2015-05-18:
Currently the following tests pass:

* publisher_subscriber_cpp__empty
  * [x] rmw_connext_cpp__rmw_connext_dynamic_cpp
  * [x] rmw_connext_cpp__rmw_opensplice_cpp
  * [x] rmw_connext_cpp
  * [x] rmw_connext_dynamic_cpp__rmw_connext_cpp
  * [ ] rmw_connext_dynamic_cpp__rmw_opensplice_cpp
  * [x] rmw_connext_dynamic_cpp
  * [x] rmw_opensplice_cpp__rmw_connext_cpp
  * [x] rmw_opensplice_cpp__rmw_connext_dynamic_cpp
  * [x] rmw_opensplice_cpp
* publisher_subscriber_cpp__primitives
  * [ ] rmw_connext_cpp__rmw_connext_dynamic_cpp
  * [x] rmw_connext_cpp__rmw_opensplice_cpp
  * [ ] rmw_connext_cpp
  * [ ] rmw_connext_dynamic_cpp__rmw_connext_cpp
  * [ ] rmw_connext_dynamic_cpp__rmw_opensplice_cpp
  * [ ] rmw_connext_dynamic_cpp
  * [ ] rmw_opensplice_cpp__rmw_connext_cpp
  * [ ] rmw_opensplice_cpp__rmw_connext_dynamic_cpp
  * [ ] rmw_opensplice_cpp
* requester_replier_cpp__emptyservice
  * [ ] rmw_connext_cpp__rmw_connext_dynamic_cpp
  * [x] rmw_connext_cpp
  * [ ] rmw_connext_dynamic_cpp__rmw_connext_cpp
  * [ ] rmw_connext_dynamic_cpp
  * [x] rmw_opensplice_cpp
* requester_replier_cpp__primitivesservice
  * [ ] rmw_connext_cpp__rmw_connext_dynamic_cpp
  * [ ] rmw_connext_cpp
  * [ ] rmw_connext_dynamic_cpp__rmw_connext_cpp
  * [ ] rmw_connext_dynamic_cpp
  * [ ] rmw_opensplice_cpp

Known limitations:
* services don't work across vendors

Due to the amount of failing tests I won't add more message / service types for now but look into why these are failing.

dirk-thomas, 2015-05-20:
Since now all tests pass for empty / all primitives messages / services I will continue adding for types.

dirk-thomas, 2015-05-20:
Currently the following tests pass:

* publisher_subscriber_cpp__staticarrayprimitives
  * [ ] rmw_connext_cpp__rmw_connext_dynamic_cpp
  * [ ] rmw_connext_cpp__rmw_opensplice_cpp
  * [ ] rmw_connext_cpp
  * [ ] rmw_connext_dynamic_cpp__rmw_connext_cpp (ros2/rmw_connext#34)
  * [ ] rmw_connext_dynamic_cpp__rmw_opensplice_cpp
  * [ ] rmw_connext_dynamic_cpp
  * [ ] rmw_opensplice_cpp__rmw_connext_cpp
  * [ ] rmw_opensplice_cpp__rmw_connext_dynamic_cpp
  * [ ] rmw_opensplice_cpp

dirk-thomas, 2015-05-26:
This should finally cover all the above tests. If nobody has suggestions for further communication tests I would propose to close this.

tfoote, 2015-05-26:
looks complete

closed 2015-05-26 by esteve