ros2/ros2 #43: Review API of generated message code

gerkey, 2015-04-21:
Revisit generated message code, make changes, document the result.

dirk-thomas, 2015-07-15:
The C++ API for messages has been documented in the article http://design.ros2.org/articles/generated_interfaces_cpp.html

I assume that by agreeing to that article everyone agrees to the current API? Is there anything else to do for this ticket?

wjwwood, 2015-07-15:
lgtm, I think this can be closed, though I expect the amount of things that are included might increase. Like functions to determine how much space is needed to preallocate and if that's even possible or not, just as an example.

dirk-thomas, 2015-07-15:
Sure, I think the best way to go forward is to continuously update the article when the implementation evolves.

closed 2015-07-15 by dirk-thomas