ros2/ros2 #46: Parameter Review Roundup

wjwwood, 2015-04-24:
- [ ] parameter_name or just type: https://github.com/ros2/rcl_interfaces/blob/master/rcl_interfaces/msg/ParameterDescription.msg
- [ ] Response callback optionally also gets original request.
- [ ] Restructure Messages based on discussion:

GetParameters.srv
```
# A list of parameter names to get.
# TODO(wjwwood): Decide on the rules for grouping, nodes, and parameter "names" in general, then link to that.
string[] names

---

# List of values which is the same length and order as the provided names.
# If a parameter was not set yet, the value will have PARAMETER_NOT_SET as the parameter_type.
ParameterValue[] values
```

ParameterType.msg
```
uint8 PARAMETER_NOT_SET=0
uint8 BOOL_PARAMETER=1
uint8 INTEGER_PARAMETER=2
uint8 DOUBLE_PARAMETER=3
uint8 STRING_PARAMETER=4
uint8 BYTES_PARAMETER=5
```

ParameterValue.msg
```
# Used to determine which of the next *_value fields are set.
# ParameterType.PARAMETER_NOT_SET indicates that the parameter was not set
# (if gotten) or is uninitialized.
uint8 parameter_type  # wjwwood: was there a reason to not use just type here?

# "Variant" style storage of the parameter value.
bool bool_value
int64 integer_value
float64 double_value
string string_value
int8[] bytes_value
```

GetParameterTypes.srv
```
# A list of parameter names.
# TODO(wjwwood): link to parameter naming rules.
string[] parameter_names

---

# List of types which is the same length and order as the provided names.
# ParameterType.PARAMETER_NOT_SET indicates that the parameter is not currently set.
uint8[] parameter_types
```

SetParameterResult.msg
```
# A true value of the same index indicates that the parameter was set successfully.
# A false value indicates the change was rejected.
bool successful
# Reason why the setting was either successful or a failure.
# Should only be used for logging and user interfaces.
string reason
```

SetParameters.srv
```
# A list of parameters to set.
Parameter[] parameters

---

# An order list of results of the same size as the given parameters.
SetParameterResult[] results
```

SetParametersAtomically.srv
```
# A list of parameters to set.
Parameter[] parameters

---

SetParameterResult result
```

Parameter.msg
```
string name
ParameterValue value
```

ListParameters.srv
```
uint64 DEPTH_RECURSIVE=0

# The list of parameter prefixes to query.
string[] parameter_prefixes
# Relative depth from given prefixes to return.
# Use DEPTH_RECURSIVE to get the recursive parameters and prefixes for each prefix.
uint64 depth

---

# All parameters under the given prefixes.
string[] parameter_names
# All prefixes under the given prefixes.
# TODO(wjwwood): link to prefix definition and rules.
string[] parameter_prefixes
```

DescribeParameters.srv
```
# A list of parameters of which to get the description.
string[] parameter_names

---

# An ordered list of parameter descriptions of the same size as the list requested.
# If the parameter_type is set to PARAMETER_NOT_SET, then the parameter requested was not set.
ParameterDescription[] parameter_descriptions
```

ParameterDescription.msg
```
string parameter_name
uint8 parameter_type  # wjwwood: same note, seems like just type would be better.
# TODO: add additional meta data.
```

- [ ] [impl] ParameterContainer should use the generated ParameterValue.msg as storage
- [ ] [api] ParameterContainer should be renamed to avoid confusion that it only holds one value and to show that it is related to the ParameterValue.msg
- [ ] [api] the parameter API should take a node not be part of the node class.
- [ ] [api] refactor examples and design information based on our discussion:

```
auto node = ...;
auto pc = ParameterClient(node, node_name="");
auto pc = ParameterClient(node, remote_node_name="other");
auto pc = node->createParameterClient(remote_node_name="other");
pc.set...;
pc.get...;
pc.list...;
```

The handle approach:
```
auto node = ...;
auto my_node_pc = ParameterClient(node, node_name="");
auto remote_node_pc = ParameterClient(node, remote_node_name="other");
pc.set...;
pc.get...;
pc.list...;
```

The single client approach:
```
auto node = ...;
auto pc = ParameterClient(node);
pc.searchForAll('*/foo/*')
pc.setP(fqpn)
pc.setP(parameter_name, node_name)
pc.setP(parameter_name)
pc.setLocalP(parameter_name)
pc.set...(node_name, ...);
pc.set...('parameter://node.ping/foo/bar');
pc.get...;
pc.list...;

AsyncParameterClient apc = pc->get_async...

auto node = Node(parameter_server=true)
auto ps = node->getParameterServer();

```

- [ ] [api] Draft (header/examples) of user's Server side of the parameters.

dirk-thomas, 2015-04-28:
After re-reading our summary from the last meeting I would suggest the following changes:

* `ParameterType.msg`: rename the enum types to exact types, e.g. `INT64` vs `INTEGER` to clarify what exact value range they can store and keeping the option to add further primitive types in the future.

* instead of the term `Description` (which could also refer to textual description) change it to `Descriptor` which is a common term for a class containing meta information like this.

wjwwood, 2015-04-28:
> ParameterType.msg: rename the enum types to exact types, e.g. INT64 vs INTEGER to clarify what exact value range they can store and keeping the option to add further primitive types in the future.

+0, the `int` in most programming languages doesn't describe the exact size of the integer. So I think there is precedence for this, but I'm not passionate about it one way or the other.

> instead of the term Description (which could also refer to textual description) change it to Descriptor which is a common term for a class containing meta information like this.

+1 That sounds good to me.

dirk-thomas, 2015-04-28:
The current `ParameterEvent` message uses `ParameterDescriptor` for the deleted parameters. Should we change this to just use the `Parameter` type?

dirk-thomas, 2015-04-28:
One message definition is still called `ParameterEventDescriptions` and should probably be named `ParameterEventDescriptors`?

dirk-thomas, 2015-04-28:
Some comments in the message definitions need to be updated:
* https://github.com/ros2/rcl_interfaces/blob/fe5506cfe874c91dcb183581a164712f646db2a5/rcl_interfaces/msg/Parameter.msg#L2
* `ParameterEvent`: mention constraints about entries with same names in different fields as discussed (same for `ParameterDescriptor`)

dirk-thomas, 2015-04-28:
Suggestion to change the field `parameter_names` in `GetParameterTypes` to match other services `DescribeParameters`.

tfoote, 2015-05-05:
@esteve @tfoote confirm checkboxes covered and close

esteve, 2015-05-26:
Naming: we're going to use name and type for ParameterDescriptor.

wjwwood, 2015-06-11:
@esteve ping

closed 2015-06-16 by esteve