ros2/ros2 #48: generate interface headers in subnamespace

dirk-thomas, 2015-05-05:


dirk-thomas, 2015-05-26:
As a first step I have polished all the message templates:

* the generated code states from which template file it was generated from
* make the generated code more readable
* generated code (mostly) following code style
* stripped obsolete comment and debug output

with the following commits:
* https://github.com/ros2/rosidl/commit/c03cd1f8490485586d4213222a2ef7a439131c50
* https://github.com/ros2/rosidl_dds/commit/034edfcd1cfaabccf486b731e625d19b54edbed1
* https://github.com/ros2/rmw_connext/commit/5606690271deaf4c043dab35a298e6ddabddd6df
* https://github.com/ros2/rmw_opensplice/commit/e78a44c3a3b29e261d8ed63bcb8a4ea06e65df67

dirk-thomas, 2015-05-29:
The PRs pass on the farm:
* http://54.183.26.131:8080/view/ros2/job/ros2_batch_ci_linux/123/
* http://54.183.26.131:8080/view/ros2/job/ros2_batch_ci_osx/135/ (OS X had the same tests fail before these changes)

@esteve @tfoote @wjwwood Please review the seven referenced PRs.

Please also comment on my previous question whether the generated message files should use camel case (+ `_Request` suffix for services) or lowercase/underscore naming (+ `__request` suffix).

The advantages of using the later one would be:

* consistent with the rest of the ROS 2 code (as we decided to use that as our "code style")
* collision free if two messages only differ in the case: `FooBar.msg` and `Foobar.msg` result in `foo_bar.hpp` and `foobar.hpp`

The disadvantages are:

* that the naming is different from what the users of ROS 1 are used to
* there is a "transformation" between the message name and the filename

esteve, 2015-06-01:
I like lowercase/underscore, even if the naming convention is not the same as ROS 1. I think it looks cleaner and consistent with our code.

wjwwood, 2015-06-01:
These are goals we already have:

- `.msg` files follow the Class naming scheme (camel case, alpha only)
- each `.msg` generates a file with only the data structure, in which primitive types resolve to built-in language types where possible (only including other message structures where ever possible).

Let me try to enumerate the options (c++, using the example `FooBar.msg`):

#### Option 1

We generate `<pkg_name>/msg/FooBar.hpp` which only contains the struct and additional functionality is included in separate files with suffixes, e.g. `FooBar_utils.hpp`, there is no "top level" file to include.

Example:

```
foo_pkg
└── msg
    ├── FooBar.hpp
    └── FooBar_utils.hpp
```

#### Option 2

We generate `<pkg_name>/msg/foo_bar.hpp` which only includes the struct and additional functionality which are in separate files with suffixes, e.g. `foo_bar_struct.hpp` and `foo_bar_utils.hpp`, the user just includes the `foo_bar.hpp` file most of the time.

Example:

```
foo_pkg
└── msg
    ├── foo_bar.hpp
    ├── foo_bar_struct.hpp
    └── foo_bar_utils.hpp
```

Optionally we can drop the `_struct` suffix as well as the conversion for the struct only header and generate `foo_bar.hpp`, `FooBar.hpp`, and `foo_bar_utils.hpp`.

Example with option:

```
foo_pkg
└── msg
    ├── FooBar.hpp
    ├── foo_bar.hpp
    └── foo_bar_utils.hpp
```

#### Option 3

We generate `<pkg_name>/msg/FooBar.hpp` which only includes the struct and additional functionality which are in separate files with suffixes, e.g. `FooBar_struct.hpp` and `FooBar_utils.hpp`, the user just includes the `FooBar.hpp` file most of the time.

Example:

```
foo_pkg
└── msg
    ├── FooBar_struct.hpp
    ├── FooBar_utils.hpp
    └── FooBar.hpp
```

wjwwood, 2015-06-01:
I think those are the options, please edit my post if you have corrections or something to add (options or goals).

We should go over these together, in person, and then decide on one of them.

dirk-thomas, 2015-06-01:
We decided to pick the following (first part of option 2):

* camel case for the message files since they define a data structure
* lower case / underscore for the generated C++ headers to follow the code style
* `<msg_name>.hpp` includes `<msg_name>__struct.hpp` and in the future also `<msg_name>__<suffix>.hpp` where suffix could be anything without additional dependencies
* `<msg_name>_<fancy>.hpp` can provide fancy features which require additional dependencies but are not included by default from `<msg_name>.hpp`

dirk-thomas, 2015-06-02:
I have added a commit to all PRs which implement the new naming of the generated message files. For C++ I would think this ticket is complete but I haven't touched C yet.

Current the C files are named:
* `MsgName-c.h`
* `MsgName_Struct-c.h`

We should clearly apply the same naming rules: lowercase / underscore, suffix `__struct`.

My question would be about the `-c` suffix. Should we just drop it now that the files are clearly separated by their extension (`.hpp` vs. `.h`, `.cpp` vs. `.c`)? Or would that be too confusing for users?

dirk-thomas, 2015-06-02:
I updated the message generation for C accordingly (dropping the `-c` suffix).

CI jobs look good (http://54.183.26.131:8080/view/ros2/job/ros2_batch_ci_linux/135/ http://54.183.26.131:8080/view/ros2/job/ros2_batch_ci_osx/148/).

@esteve @tfoote @wjwwood Any final comments on the updated PRs before (squashing and) merging?

esteve, 2015-06-02:
Dropping the `-c` and relying on the extension sounds good to me, though I agree it may be confusing. Should we look into putting the C and C++ generated files into separate directories?

dirk-thomas, 2015-06-02:
A language specific part in the folder name doesn't sound appealing to me. Do you have any specific suggestions?

wjwwood, 2015-06-02:
+1 on the pull requests.

dirk-thomas, 2015-06-02:
We talked about it and will keep it as it is for now. If it turns out to be an issue we can move the C code into a differently named folder.

tfoote, 2015-06-02:
I think that making sure to have good documentation of the extensions and relying on them would be best for now. It is guarenteed to be unique, and we want to have the most native experience which means that you don't self reference your language.

closed 2015-06-02 by dirk-thomas