ros2/ros2 #53: eliminate warnings

dirk-thomas, 2015-06-02:


wjwwood, 2015-06-15:
There are no warnings on Linux atm, and only 3 on OS X, for which it is not clear how I should fix it just yet. Windows still has a lot of warnings, but I just got it rebuilding and tracking the warnings.

I propose we merge the current pull requests as is and push this issue back to the backlog in favor of other tasks. At least now we've eliminated the most crucial warnings and now we're tracking the warnings so we'll know if we've introduced a bunch all at once.

So, if everyone could give a +1/-1 for my suggestion I'd appreciate it.

dirk-thomas, 2015-06-15:
The lastest state of some of the pending PRs is still that they should be evaluated on Windows. Can you add links to CI builds for them to quantify their usefulness?

wjwwood, 2015-06-15:
Well they build on Windows, which was the concern, so that's not a problem. I haven't audited Windows for warnings. Here are the latest CI jobs:

- Linux: http://54.183.26.131:8080/job/ros2_batch_ci_linux/145/ (0 warnings)
- OS X: http://54.183.26.131:8080/job/ros2_batch_ci_osx/156/ (3 warnings)
- Windows: http://54.183.26.131:8080/job/ros2_batch_ci_windows/286/ (?)

The windows job didn't have the "fix idl" branch merged in, so I just reran it, but locally it builds but there are a lot of warnings. I'll update this comment once the windows job finishes.

wjwwood, 2015-06-15:
Actually it looks like the Windows build is now broken? Maybe this is related to the recent changes with the PYTHONPATH and PATH?

http://54.183.26.131:8080/job/ros2_batch_ci_windows/286/console
```
+++ Building 'ament_cmake_core'
Running cmake because arguments have changed.
==> 'C:\Jenkins\workspace\ros2_batch_ci_windows\workspace\build\ament_cmake_core\cmake__build.bat C:\Program Files (x86)\CMake\bin\cmake.EXE C:\Jenkins\workspace\ros2_batch_ci_windows\workspace\src\ament\ament_cmake\ament_cmake_core -DAMENT_ENABLE_TESTING=1 -DCMAKE_INSTALL_PREFIX=C:\Jenkins\workspace\ros2_batch_ci_windows\workspace\install -G Visual Studio 14 2015 Win64' in 'C:\Jenkins\workspace\ros2_batch_ci_windows\workspace\build\ament_cmake_core'
-- Found PythonInterp: C:/tools/python/python.exe (found suitable version "3.4.3", minimum required is "3") 
-- Using PYTHON_EXECUTABLE: C:/tools/python/python.exe
ImportError: 'from ament_package import parse_package_string' failed: No module named 'ament_package'

Make sure that you have installed 'ament_package', it is up to date and on the PYTHONPATH.

CMake Error at cmake/core/ament_package_xml.cmake:94 (message):
  execute_process(C:/tools/python/python.exe
  C:/Jenkins/workspace/ros2_batch_ci_windows/workspace/src/ament/ament_cmake/ament_cmake_core/cmake/core/package_xml_2_cmake.py
  C:/Jenkins/workspace/ros2_batch_ci_windows/workspace/src/ament/ament_cmake/ament_cmake_core/package.xml
  C:/Jenkins/workspace/ros2_batch_ci_windows/workspace/build/ament_cmake_core/ament_cmake_core/package.cmake)
  returned error code 1
Call Stack (most recent call first):
  cmake/core/ament_package_xml.cmake:49 (_ament_package_xml)
  CMakeLists.txt:13 (ament_package_xml)

-- Configuring incomplete, errors occurred!
See also "C:/Jenkins/workspace/ros2_batch_ci_windows/workspace/build/ament_cmake_core/CMakeFiles/CMakeOutput.log".

<== Command 'C:\Jenkins\workspace\ros2_batch_ci_windows\workspace\build\ament_cmake_core\cmake__build.bat C:\Program Files (x86)\CMake\bin\cmake.EXE C:\Jenkins\workspace\ros2_batch_ci_windows\workspace\src\ament\ament_cmake\ament_cmake_core -DAMENT_ENABLE_TESTING=1 -DCMAKE_INSTALL_PREFIX=C:\Jenkins\workspace\ros2_batch_ci_windows\workspace\install -G Visual Studio 14 2015 Win64' failed with exit code '1'
<== 'env.bat C:\tools\python\python.exe -u ".\src\ament\ament_tools\scripts\ament.py" build --build-tests --build-space "build" --install-space "install" "src"' exited with return code '1'
```

dirk-thomas, 2015-06-15:
It looks like Windows is choking on the arguments defining the environment hooks (which are now two separated by a semicolon instead of one). I committed the following change which should help: ament/ament_tools@1fe24740292581a9f8f40c1bddb8aad00955b919

wjwwood, 2015-06-15:
Ok, I'll rerun it.

dirk-thomas, 2015-06-16:
+1 to merge the referenced PRs since they do improve the current state.

wjwwood, 2015-06-16:
bump @esteve @tfoote @jacquelinekay for +1/-1

tfoote, 2015-06-16:
+1 merge and rereview the state afterwards.

esteve, 2015-06-16:
+1 merge

wjwwood, 2015-06-16:
Thanks, I'll go ahead and merge the PR's and then move this issue back to the backlog for now.

wjwwood, 2015-06-16:
Ok, I've done that and update the description of this issue to capture what's left todo.

dirk-thomas, 2015-07-27:
Recent PRs on this topic:
* ros2/rmw#24
* ros2/rmw_connext#61
* ros2/rmw_opensplice#56

This gets us to no warnings on Linux and OS X (again).
The remaining Windows warnings are:
* either type conversion warnings
* or usage of deprecated functions (due to no upper limit for string processing)

dirk-thomas, 2015-07-30:
All warnings have been resolved except:

* usage of `strerror`
* usage of `strcpy`
* usage of `_snprintf`

dirk-thomas, 2015-08-17:
After the latest set of PRs their is only one kind of warnings left: http://ci.ros2.org/job/ros2_batch_ci_windows/201/warnings34Result/category.63481637/package.-2136935613/

@esteve Can you please fix those so that we can enable the email notification for the nightly Windows job.

esteve, 2015-08-17:
@dirk-thomas 

PR: https://github.com/ros2/demos/pull/9
CI job: http://ci.ros2.org/job/ros2_batch_ci_windows/210/

dirk-thomas, 2015-08-17:
I think the remaining 10 warnings are related to the failing tests: http://ci.ros2.org/job/ros2_batch_ci_windows/213/warnings34Result/

dirk-thomas, 2015-08-19:
Last warning fixed (suppressed) by ros2/ros2#90.

closed 2015-08-19 by dirk-thomas