ros2/ros2 #54: [demo] Camera images via lossy wifi

wjwwood, 2015-06-11:
Goals:

- [ ] Demonstrate better comms over lossy wifi with large message, e.g. Camera Images
- [ ] Provide a method to try out different QoS settings (similar DDS's shapes demo)

Risks:

- How to emulate bad network?
 - Maybe use `tc`?

esteve, 2015-06-15:
Add QoS options to publisher/subscriber APIs

wjwwood, 2015-06-16:
Multiple streams with different QoS, should stress network.
Multiple OS in the demo.

BobDean, 2015-06-29:
what resolution images and at what Hz are you targeting? My default question for people is "How will you handle a 25MPixel stereo pair"?  

This NIST link may help track down a network simujlator: http://www-x.antd.nist.gov/nistnet/

wjwwood, 2015-06-29:
We were going to start with something more reasonable, like 1280x1024 at 30 hz. I don't think it's realistic to do 25Mp stereo pair over crowded wifi, but we can work up to it in our testing. The 25Mp pair might be a valid test for the intraprocess image pipeline, but that really depends on the computer you're running it on.

NIST net looks like an interesting lead on network simulation. Thanks @BobDean.

gerkey, 2015-08-10:
We now have a test setup working, publishing images from camera to a camera viewer node.

Next up is tweaking QoS settings, and if necessary, messing with network interface behavior via `tc`.

Write a single-host tutorial that shows how to see the effect over localhost.  Might need to disable shared memory transport somehow.

dirk-thomas, 2015-09-08:
Shipped.

closed 2015-09-08 by dirk-thomas