ros2/ros2 #55: [demo] Image Processing Pipeline

wjwwood, 2015-06-11:
Goals:

- [ ] Demonstrate use of efficient intra-process comms
- [ ] Demonstrate configuration time organization of nodes into process
- [ ] Demonstrate using non-ROS types in comms  (type masquerading)

wjwwood, 2015-06-15:
@dirk-thomas @jacquelinekay @tfoote @esteve Can you guys look over this updated description and mention anything you think is missing or unclear?

jacquelinekay, 2015-06-15:
Can you concretely the magnitude of "large messages" and "high rates" in the first paragraph of the description?

jacquelinekay, 2015-06-15:
Other considerations, I don't know how specific you want to get:

Provide a concrete metric for when each feature will be considered done for the demo. For example, the goal for type masquerading could be "show efficient transport of an OpenCV image masquerading as a generic Image type in both intra and inter-process communication".

Give a dependency graph for this set of features (unless you think this is already adequately captured in the MindMup)

dirk-thomas, 2015-06-15:
Looks detailed and complete to me.

wjwwood, 2015-06-15:
> Can you define the magnitude of "large messages" and "high rates" in the first paragraph of the description?

That's a good idea, I'll come up with something that is typical, e.g. 30 Hz and 1280x1024 pixel RGB images.

> Provide a concrete metric for when each feature will be considered done for the demo. For example, the goal for type masquerading could be "show efficient transport of an OpenCV image masquerading as a generic Image type in both intra and inter-process communication".

I can try to go into more detail about the acceptance criteria, but I wasn't sure exactly what type I would use, so I left details like using OpenCV's Mat object or something like that out of the description. The point about type masquerading working in both intra and inter process comms is valid, I'll add that.

> Give a dependency graph for this set of features (unless you think this is already adequately captured in the MindMup)

I think these features are actually (maybe surprisingly) isolated from each other in terms of dependencies. The sub bullets of the node component model stuff is fairly sequential, but ideally they can be done independently of intra-process, type masquerading, and component life cycle items.

dirk-thomas, 2015-08-19:
I pushed the forked `pluginlib` which uses the forked `class_loader` as well as the ament index through `ament_index_cpp`: https://github.com/ros/pluginlib/tree/ros2

wjwwood, 2015-09-08:
TODO for me. Summarize what's been done for the alpha 1 and move remaining things into new issues.

wjwwood, 2015-09-16:
Ok, I've struck out anything we didn't close above, and I spun out a few issues for some of the more immediate tasks. We only managed to accomplish part of this demo:

- We got intra process working.
- We did not get loading of nodes by components done.
- We did not get Type Masquerading implemented.
- We did not get to extending launch with syntactic sugar for launching nodes remotely.

closed 2015-09-16 by wjwwood

dhood, 2016-11-04:
For anyone that comes across this ticket: see https://github.com/ros2/ros2/wiki/Intra-Process-Communication for the demo and https://github.com/ros2/demos/tree/release-alpha8/intra_process_demo for the accompanying code