ros2/ros2 #56: [demo] Visualization of Data with ROS 1 tools

wjwwood, 2015-06-11:
Goals:

- [ ] Bidirectional pub/sub with ROS 1
- [ ] Visualize data with rqt/rviz
- [ ] Send commands from ROS 1 to ROS 2, e.g. teleop

dirk-thomas, 2015-06-15:
The purpose of this demo is to demonstrate the bidirectional exchange of information between a ROS 1 and ROS 2 system.
It is the foundation to gradually migrate existing applications from ROS 1 to ROS 2.
It will also enable us to use ROS 1 tools until similar tools become available in ROS 2, e.g. `rosbag`, `rqt`.

Use case in the demo:

* visualize data coming from a ROS 2 componenent with ROS 1 `rqt` / `rviz`
* send commands from ROS 1 to ROS 2, e.g. teleop

### Exchanged information

* **messages:** publish / subscribe across ROS versions
* **services:** request / reply across ROS versions
* **actions:** request / progress / reply / preemption across ROS versions
* **parameters:** access from ROS 2 to ROS 1 global parameters
* **dynamic reconfigure:** (?)

The demo will only make use of messages.

### Required meta information / introspection / functionality

* List available message / service types

  In ROS 1 we can use the `rosmsg` API to find them.
  In ROS 2 we can utilize the ament resource index to find them.

* List available topics / servers with their types at runtime
* List available servers with their types at runtime

  For ROS 1 these information can be polled from the `rosmaster`.
  For ROS 2 there is currently no API to provide these information.

* Bidirectional mapping between ROS 1 and ROS 2 data types

### Technical requirements

* [ ] Messages / services be must register at the ament resource index

* [ ] Compile / link bridge code against ROS 1 and ROS 2 API at the same time

  The bridge could be:
  * a standalone process
  * a ROS 1 nodelet in a `catkin` package
  * a ROS 2 component in an `ament` package

* [ ] Rules for mapping package names, message names, primitive types and field names between ROS 1 and ROS 2
  * Simple (almost 1-to-1) mapping
  * Allow custom mapping (?)

  The set of bridgeable types could be based on the message / service definitions available:
  * at compile time of the bridge
  * at runtime time of the bridge (do we think this is feasible with reasonable effort?)

* [ ] ROS 2 graph API and / or events about changes of the ROS graph

### Questions to clarify

* How are `catkin` and `ament` workspaces used at the same time to create a "dual-homed" binary?

* We have almost no ROS 2 message definitions atm. We could "convert" a set of message packages into ROS 2 packages? If yes, which ones?

* How can we specify mapping rules beside the simple (almost) 1-to-1 mapping of package names, message names, primitive types and field names?

dirk-thomas, 2015-06-23:
* [x] create a unidirectional bridge with a single hard coded topic / type: ros2/ros2#65
* [ ] create a bidirectional bridge with a single hard coded topic / type: ros2/ros1_bridge#1
* [ ] support arbitrary message types available at compile time of the bridge
* [ ] establish bridging based on available topics at runtime

gerkey, 2015-07-14:
We're currently blocked on how to best get current topic name and type information on the ROS2 side.  For now, hardcoding that information in the bridge works fine.  So the next step will be to create a ROS2 publisher (e.g., image from webcam) and show it in rviz.  The demo gets better later when the bridge doesn't need the hardcoded info.

dirk-thomas, 2015-07-18:
Please see the update on the current state of the demo in the update previous comment.

dirk-thomas, 2015-07-20:
I have created a few more PRs which makes the demo "work" in all cases. The only drawback currently is that when exchanging images between Connext and OpenSplice the performance is really bad. Only some images arrive at all and they also have a long delay...

dirk-thomas, 2015-07-28:
As a back channel I will add a simple toggle topic to enable / disable a specific OpenCV function applied on the image sender side.

dirk-thomas, 2015-07-30:
I added a topic as a backchannel from rqt to the ROS 2 cam2image node: ros2/demos#2

gerkey, 2015-08-10:
Demo is basically working (modulo noticing tear-down of topics on the ROS 2 side).  Next up: write documentation for how try the demo (tutorial with screenshots).  Add to the README.

dirk-thomas, 2015-09-04:
Shipped :ship:

closed 2015-09-04 by dirk-thomas