ros2/ros2 #58: [demo] Real-time node

wjwwood, 2015-06-11:
Goals:

- [ ] Node using real-time process to run a PID control loop
- [ ] Demonstrate using life cycle to control memory allocation
- [ ] Demonstrate external pub/sub does not affect real-time thread

jacquelinekay, 2015-06-18:
do we need to implement an "rcl_realtime" library for this demo, or patch `rclcpp` to be realtime-safe, or implement `rclc` to be realtime-safe?

similar question for `rmw`

wjwwood, 2015-06-18:
Ideally we'd have one library this is capable of both, with either no changes or only configuration changes to achieve real-time. However, if that's not possible then we'll have to consider what our options are. If we have to provide an entirely separate library for real-time, then it's hard to say how that's better than just having good integration with something like OROCOS.

jacquelinekay, 2015-06-18:
Cool. From what I've learned today, I think it's reasonable to have one library which can be configured for realtime-safe behavior. [This example for RT Preempt](https://rt.wiki.kernel.org/index.php/Threaded_RT-application_with_memory_locking_and_stack_handling_example) shows how to preallocate the stack and lock the memory for a process so that starting new threads and dynamic memory allocation will not result in pagefaults. After that, the challenge is figuring out how much memory to preallocate.

Not sure how many platforms this solution will work on, but I think it will work fine for the demo...

closed 2015-09-08 by jacquelinekay