ros2/ros2 #61: Setup RT Linux Machine and Example RT program

wjwwood, 2015-06-16:


jacquelinekay, 2015-06-22:
I'm taking this over because it's a prereq for my card (#62).

jacquelinekay, 2015-06-23:
The machine is set up (thanks to the stable RT Preempt kernel maintained by IHMC). What are the requirements for an example RT program? [This example](https://rt.wiki.kernel.org/index.php/Squarewave-example) from the RTLinux wiki has some nice stuff going on (continuous real time loop) but its output has to be measured with an oscilloscope, which is not great

gerkey, 2015-07-07:
Working on DDS-only examples to see what can be done with real-time constraints.  So far, OpenSplice is showing many page faults in common usage (at least in the publisher).  Next up is trying Connext, and consider using custom allocator with OpenSplice.

If neither approach can satisfy real-time constraints, then it means that we need to short-circuit the DDS publish calls for intraprocess comms, as well as ensure that if there's an interprocess subscriber, that we don't introduce a violation.

gerkey, 2015-07-07:
Whenever it's convenient, submit a pull request with your evolving examples for review and feedback.

closed 2015-07-15 by jacquelinekay