ros2/ros2 #62: Determine if DDS can do intra-process pub/sub in real time

wjwwood, 2015-06-16:


wjwwood, 2015-06-16:
Both research and test it.

tfoote, 2015-06-23:
We want to know if we can build realtime safe ROS intraprocess communications on top of DDS message passing. (It may depend on DDS implementation)

jacquelinekay, 2015-07-01:
- [x] Complete simple intra-process pub/sub over example (https://github.com/osrf/ros_dds/blob/large_message_rt/realtime/dds_intraproc.cpp)
- [ ] complete instrumentation of example and show plots with results verifying real-time performance (or not)

jacquelinekay, 2015-07-09:
test library:
https://github.com/jacquelinekay/rttest/pull/1

examples:
https://github.com/osrf/ros_dds/pull/27
https://github.com/osrf/ros_dds/pull/26
https://github.com/osrf/ros_dds/pull/28

closed 2015-07-15 by jacquelinekay