ros2/ros2 #66: Add additional QoS settings to Pub/Sub API

wjwwood, 2015-06-16:


esteve, 2015-06-17:
We can provide a limited set of options (reliable vs best effort) via the API, but allow the users to set more complex QoS policies via an external file.

dirk-thomas, 2015-07-30:
I am unsure if the PRs are ready to be reviewed. Should all the referenced PRs have the label "in review" or should this ticket have the label "in progress"?

esteve, 2015-07-30:
I'm going to put all the PRs in review, so that I can move forward. Since the changes I made affect the API, every time there's a conflicting change I spend too much time rebasing. I'll add the review tag to all the PRs to mark them as ready. Thanks for noticing!

dirk-thomas, 2015-08-04:
Add the proposed policies to the design document: ros2/design#36

dirk-thomas, 2015-08-04:
Can you please run the CI jobs and link them here? The last referenced job in one of the PRs had regressions.

esteve, 2015-08-05:
Linux CI job:

http://ci.ros2.org/job/ros2_batch_ci_linux/156/

Mac OSX job:

http://ci.ros2.org/job/ros2_batch_ci_osx/84/

The Windows slave is pending because of the VS2015 upgrade.

esteve, 2015-08-06:
I fixed the syntax style, the linked PR should be ready to merge. I'd rather not wait for the Windows slave to be ready and fix any Windows issues afterwards.

dirk-thomas, 2015-08-06:
ros2/demos#4 is currently not mergable.

esteve, 2015-08-06:
@dirk-thomas I just fixed it, be a little patient.

esteve, 2015-08-07:
Latest CI jobs:

http://ci.ros2.org/job/ros2_batch_ci_linux/163/

http://ci.ros2.org/job/ros2_batch_ci_osx/91/

esteve, 2015-08-07:
Linux and Mac OSX CI jobs are green, going to merge the linked PRs.

closed 2015-08-07 by esteve

dirk-thomas, 2015-08-10:
The ros1_bridge package (https://github.com/ros2/ros1_bridge) needs the same updates to be continue to compile. I will create a patch for it since I need to compile the package for ros2/ros1_bridge#3.