ros2/ros2 #67: Setup ROS 2 Specific Jenkins Master

wjwwood, 2015-06-22:
Really, the only thing that has to be done is to update Jenkins to the latest version because that fixes the issue with the VS 2015 license not updating on Windows. It would just be easier to do so if it were not tied to the experimental ROS 1 build farm.

@tfoote and I discussed simply getting one EC2 machine that's both the Jenkins Master as well as the ROS 2 Linux build slave. The Windows and OS X build slaves are at my desk.

It's not clear to me whether or not we should fork the ROS 1 build farm deployment configurations or just set one up temporarily, documenting the setup in a wiki page or something until we need a more sophisticated build farm.

tfoote, 2015-06-22:
Yeah, I think separating them makes sense. Until we setup the rosdistro based building we don't need the fully provisioning.

tfoote, 2015-06-23:
A machine for this has been stood up at: 52.8.225.206

wjwwood, 2015-06-23:
Ok, I can reach it. Thanks.

wjwwood, 2015-06-25:
Ok, the new farm is up and running:

http://ci.ros2.org/

Let me know about any issues you run into.

closed 2015-06-25 by wjwwood