ros2/ros2 #68: Refactor for real-time code safety

jacquelinekay, 2015-06-23:


jacquelinekay, 2015-06-23:
To be revisited after William submits in review work to refactor `rmw_connext` and `rmw_opensplice`

jacquelinekay, 2015-06-23:
bash script to reproduce the markdown table:

https://gist.github.com/jacquelinekay/292d010fbdfe67254ad7

Not sure which repo this belongs in.

adolfo-rt, 2015-06-30:
These results report usage of non real-time safe calls, anywhere. What really matters is if (and how) they are used in the real-time code path. For instance, if a `std::vector` is pre-allocated in the non-rt path, and used without resizing in the rt path, then nothing's wrong.

jacquelinekay, 2015-07-09:
- [ ] Find an alternative to `malloc` and `free` calls in `Executor::wait_for_work`
- [ ] Pass timeout parameter to `rmw_wait` and expose it to the Executor interface, to avoid up to `
- [ ]

closed 2015-09-08 by jacquelinekay