ros2/ros2 #69: Fix tear down issues

wjwwood, 2015-06-25:
Through out the rmw implementations there are missing destroy functions and plenty of places in rclcpp where they are not being used as a result.

This has been causing issues for me when testing out my prototype for the intra-process comms. So I've been working on fixing it. I'll open some pull requests soon and attach them to this issue.

wjwwood, 2015-06-30:
I've finished my preliminary pull requests (which are connected to this one) but I still need to run the style tool over them and run the tests. However, they are big change sets and the general ideas are done. Please start to have a look at them as soon as possible. I'll try to get the style checks out of the way tonight to comments on those.

wjwwood, 2015-06-30:
I fixed any outstanding style issues according to `ament_uncrustify`. I'll run the tests in the morning and the examples too. Those will likely produce some new errors, caught and uncaught, which are a result of these changes.

jacquelinekay, 2015-06-30:
Can we add some tests to explicitly check proper teardown in these libraries, and/or a regression test that reproduces the blocking in teardown issue you mentioned in [#51](https://github.com/ros2/rclcpp/pull/51)?

wjwwood, 2015-06-30:
> Can we add some tests to explicitly check proper teardown in these libraries, and/or a regression test that reproduces the blocking in teardown issue you mentioned in #51?

I think the current tests will test most of the code paths I've changed here. But I can look at adding a test that reproduces the segfault's I was getting with my prototype.

wjwwood, 2015-06-30:
Linux is now building, but has test failures (I'm still looking into those locally):

http://ci.ros2.org/job/ros2_batch_ci_linux/14/

OS X is the same:

http://ci.ros2.org/job/ros2_batch_ci_osx/10/

Windows remains broken, because I haven't had time to look at the current problem.

wjwwood, 2015-07-15:
Ok, I've rebased and run uncrustify over the pr's.

There are still a few failing tests, but they all seem to be related to not being able to register the type in connext_dynamic when a static or dynamic array are involved.

I'll have a look at it tomorrow with @dirk-thomas, but I don't expect any earth shattering changes to come from it.

There are quite a few things that remain to be done, but I'm going to focus on fixing the tests and then make tickets for the others lingering items. So if you want to make a final review of the pr's please do so now. Thanks.

dirk-thomas, 2015-07-15:
See https://github.com/ros2/rmw_connext/pull/45#issuecomment-121690151 for the fix of the rmw_connext_dynamic_cpp PR.

wjwwood, 2015-07-15:
Thanks for that fix @dirk-thomas.

I have squashed all the branches and so I've started a new CI job for each OS:

- Linux: http://ci.ros2.org/job/ros2_batch_ci_linux/44/
- OS X: http://ci.ros2.org/job/ros2_batch_ci_osx/21/
- Windows: http://ci.ros2.org/job/ros2_batch_ci_windows/14/ (has known other failure)

Hopefully this will reflect that OS X now passes all tests as is the case on local machine, which should also close: https://github.com/ros2/system_tests/issues/8

wjwwood, 2015-07-15:
The OS X job failed all of the RTI tests, but I applied the fix from https://community.rti.com/kb/osx510, as the error suggested, and we'll see if this new job does better:

http://ci.ros2.org/job/ros2_batch_ci_osx/22/

wjwwood, 2015-07-15:
The OS X job has passed all tests, I'll merge and close/update other tickets.

dirk-thomas, 2015-07-15:
@wjwwood Can you please add the kb reference about OS X to the example wiki page.

wjwwood, 2015-07-15:
@dirk-thomas it's on the Jenkins Slave setup wiki page (internal.local). I can add a note about it, but it isn't always necessary, maybe hardware dependent or something because I didn't have to do it on my machine.

closed 2015-07-16 by wjwwood