ros2/ros2 #70: switch to Connext 5.2

dirk-thomas, 2015-07-18:
* [ ] depends on the source being available to us
* [ ] create new Debian package
* [ ] create VS 2015 package: ros2/ros2#51
* [x] update rmw_connext repo: ros2/rmw_connext#54
* [x] update tests to cover unbounded string and sequences: ros2/system_tests#17

esteve, 2015-07-24:
Preliminary Debian packaging in:

https://github.com/rticommunity/rticonnextdds-src/pull/8

Debian packages in:

https://github.com/osrf/rticonnextdds-src/tree/binaries/ubuntu/trusty/amd64/5.2.0-0

dirk-thomas, 2015-07-28:
PR to update the Docker image: ros2/ros2#75

esteve, 2015-07-28:
Final Debian packaging already merged into upstream RTI repo:

https://github.com/rticommunity/rticonnextdds-src/commit/13830c921718cdbef801179d14ba8b53cc83a18a

Debian packages in:

https://github.com/osrf/rticonnextdds-src/tree/binaries/ubuntu/trusty/amd64/5.2.0-1

CI scripts:

https://github.com/ros2/ros2/pull/75

dirk-thomas, 2015-08-15:
I am closing this since the only remaining task is the custom Connext package built with Visual Studio 2015 which is tracked in a separate ticket.

closed 2015-08-15 by dirk-thomas