ros2/ros2 #71: raise warning level

dirk-thomas, 2015-07-20:
I have create a `raise_warning_level` branch on various repositories to raise the warning level. For now it includes: `-Wall -Wextra` Please check the changes in those branches. I would like to avoid creating separate PRs for all of those.
* https://github.com/ros2/common_interfaces/tree/raise_warning_level
* https://github.com/ros2/examples/tree/raise_warning_level
* https://github.com/ros2/rcl_interfaces/tree/raise_warning_level
* https://github.com/ros2/rclcpp/tree/raise_warning_level
* https://github.com/ros2/rmw/tree/raise_warning_level
* https://github.com/ros2/rmw_connext/tree/raise_warning_level
* https://github.com/ros2/rmw_opensplice/tree/raise_warning_level
* https://github.com/ros2/rosidl/tree/raise_warning_level
* https://github.com/ros2/rosidl_dds/tree/raise_warning_level
* https://github.com/ros2/system_tests/tree/raise_warning_level

The CI build is happy with the stricter settings: http://ci.ros2.org/job/ros2_batch_ci_linux/58/

wjwwood, 2015-07-21:
Ok, lgtm.

I'm not sure I would have added `-Wextra`, since the most prominent changes throughout are about the unused parameters which are imo false positives since it is supposed to be detecting unintentionally unused parameters, but in all of these cases the parameters are intentionally unused because they're generated code or part of an interface. Since the compiler cannot know our intention I think it's a bad check to have. Also, avoiding this warning led you to make lots of extra conditional code in our generators to avoid that, making it harder to read and understand, of which I'm not really a fan. Did you consider just doing something like `(void)unused_parameter;  // Intentionally unused.` in the function body instead? That all being said, you can go ahead with merging, it's not worth the effort to update all the branches at this point.

Also you should really open pull requests, since its basically impossible to connect the comments I made on the commits to this issue. But failing that you should at least link to the comparisons rather then making each of us figure out which commits were different from master.

dirk-thomas, 2015-07-21:
I disagree with your opinion that the *unused parameter* check is not helpful. A very good example is actually in one of the patches (https://github.com/ros2/rmw_connext/commit/81ffa776045deb9ab2248c7ea1c71c52d7d14343#diff-becdf01440f1757df19f4775b534e066R820) where I actually fixed a significant bug in the code which I only discovered because of the new warning.

I will update the commits to keep the variable if it is part of an API which should be documented in the future. In those cases I will use the `(void)var` line as suggested. In cases where the variable name is not part of an API I will keep it with the name removed.

I will create PRs for those branches. I just consider them to be a significant overhead in this case but I can stick to that in the future.

wjwwood, 2015-07-21:
Ok, that's a good point, we can leave the `-Wextra`.

wjwwood, 2015-07-21:
AFAICT, ros2/rmw_connext@81ffa77#diff-becdf01440f1757df19f4775b534e066R820 is not an unused parameter warning. Maybe I should more specifically say, `-Wextra` is probably fine, but maybe it should be accompanied by `-Wno-unused-parameter` or whatever the disabling flag is.

dirk-thomas, 2015-07-21:
The warning was in fact an unused variable since the variable allocated above was never used but instead a new allocation was done.

dirk-thomas, 2015-07-21:
Anyway I created the 10 PRs. @ all: Please review them.

wjwwood, 2015-07-21:
So, there is no warning level distinction between an unused variable and an unused parameter? I'm pretty sure clang distinguishes between the two, but I'd have to double check.

Whatever, `-Wextra` is fine.

wjwwood, 2015-07-21:
+1 to all of them.

closed 2015-07-21 by dirk-thomas