ros2/ros2 #72: Update to the official VS 2015 release

wjwwood, 2015-07-21:


j-rivero, 2015-08-05:
It should not be difficult. I was about to do it, but found a really minimal installation or configuration of Windows when connected through VPN to eatable. I can not see or touch more than the Jenkins Window. How can I interact with that? Is it possible to set it back to the standard Windows desktop?

wjwwood, 2015-08-05:
It's been like that for a while, I just restarted the machine, maybe that will fix the desktop.

j-rivero, 2015-08-05:
That made the trick, thanks William.

j-rivero, 2015-08-05:
One minor issue before installing. You are currently passing `amd64` as argument to vcvarsall.bat, which configures the 64 bit native compiler to generate 64 bit code. That is ideal, but for some reason, the free Express editions of VS don't support it. However, these Express versions support the `x86_amd64` code, which runs the 32 bit compiler to generate 64 bit code.

In our case, we can go with VS Professional and use the amd64 keyword, no problem. But if the building farm is designed to be reconfigurable, maybe it is better to use the Express version and/or the `x86_amd64`.

dirk-thomas, 2015-08-05:
I have create a branch for the CI scripts: https://github.com/ros2/ros2/tree/x86_amd64 which uses what you suggest (https://github.com/ros2/ros2/commit/3a0e537d575a042308fe99b3903c084fe70fc385)

You can try this by passing `x86_amd64` to the `CI_SCRIPTS_BRANCH` parameter of the job.

j-rivero, 2015-08-05:
Nice, let's try installing the Express version then.

j-rivero, 2015-08-05:
The update should be ready, to 2015 Express version. The system needed a reboot, a press the button but I can not contact with the machine again. Maybe is running Boot Camp and need to restart it on Windows? Hopefully it is not broken.

j-rivero, 2015-08-05:
The [testing run](http://ci.ros2.org/job/ros2_batch_ci_windows/107/console) since it can not find the devenv.exe utility. Our fork of [uncrustify](https://github.com/ament/uncrustify/blame/master/CMakeLists.txt) is calling it to [upgrade](https://msdn.microsoft.com/en-us/library/w15a82ay.aspx) .sln files.

I've verified that it is not present in the Express edition, and some people is using 'an equivalent' command, which is in the Express edition seems to be VWDExpress.exe. It has no option for "Upgrade".

Two solutions:
 * VC team is also recommended the [VCUpgrade.exe](http://blogs.msdn.com/b/vcblog/archive/2010/03/25/to-the-command-line-enthusiasts-some-quick-know-hows-for-upgrading-to-vs-2010.aspx) which was introduced in 2010 and could be done what we need. It is present in the Express edition.
 * Install the Professional version and give a look if it ship devenv.exe and it still has the upgrade option.

wjwwood, 2015-08-05:
It's strange that they removed `devenv` in the express edition. It's in the RC of VS2015, so I'd expect it to at least be in 2015 professional.

It would be nice to work with the express edition I guess, but if we need to we can step up to professional.

esteve, 2015-08-05:
Microsoft now recommends using MSBuild (which is included in VS 2015 Express) instead of Devenv:

https://msdn.microsoft.com/en-us/library/xee0c8y7.aspx

but I don't know if MSBuild can upgrade .sln files.

wjwwood, 2015-08-05:
We use `msbuild` for everything but the upgrade. It looks like it doesn't have the upgrade option though.

j-rivero, 2015-08-06:
After searching and playing for a bit, some conclusions:
 *  VCUpgrade.exe does not play with .sln files, so we can not upgrade it. I did not find any command line tool to do it.
 * The files [can easily be upgraded manually]( https://github.com/ament/uncrustify/commit/d0049a9ece6e2fb18f06438b2290c13d1b2d1725) (and cross your fingers).

Not sure what we want to do here. The fastest solution would be to check if the Professional edition ships it and go for it until we find a solution. Since the RC has it, I would expect the final version to also have it, but who knows.

The other solution would be to add some cmake logic that would be able to patch/generate the .sln file automatically and use the `VCupdater.exe` for the .vcproj, probably using the `MSVC_VERSION`, but that will require patching in the future.

wjwwood, 2015-08-06:
> The files can easily be upgraded manually (and cross your fingers).

https://github.com/ament/uncrustify/pull/6

j-rivero, 2015-08-06:
> ament/uncrustify#6

Ouch. Yes, this approach [works](http://ci.ros2.org/job/ros2_batch_ci_windows/116/consoleFull#console-section-37).

After hacking around to autogenerate the sln file (using cmake MSVC variables and parsing the msbuild version) we still need to figure out how to get the "File format version" corresponding to each version. No go for me.

We could create a couple of .sln/.vcproj files for each MSVC supported version and call the corresponding one by: relying on cmake MSVC_VERSION (note that it updates new VC version only in new cmake versions)  or parsing msbuild output (ugly).  

On the other hand, I've verified that devenv.exe is present in the Professional version.

dirk-thomas, 2015-08-06:
I think a major goal is that the ROS 2 build farm only uses "open" tools. Therefore I would argue the professional payed version of VS is not a good option.

We also don't have to support the more then a single VS version. Everything older then 2015 is not going to successfully build ROS 2 anyway. Therefore I would propose to try committing the converted VS project to the uncrustify repo and go from there with the express version of VS.

dirk-thomas, 2015-08-06:
Also please keep in mind once a new version of uncrustify is released and it becomes available in the Ubuntu release (likely 16.04) we don't even have to bother with our custom fork anymore.

j-rivero, 2015-08-07:
We hit a compilation problem, described in ros2/examples#43.

The plan is to back to the restoration point before changing the VS2015 on Aug 5th, to get back the RC compiler (14.0.22816.0). I was not able to find an installer for it. @wjwwood do you remember how did you get it?

esteve, 2015-08-07:
@j-rivero I installed the RC compiler from MSDN, but it doesn't seem to be available anymore. It'd be good to come up with a minimal and reproducible example to report it to MS.

j-rivero, 2015-08-07:
> @j-rivero I installed the RC compiler from MSDN, but it doesn't seem to be available anymore. It'd be good to come up with a minimal and reproducible example to report it to MS.

Agree, as the way for final solution. I tried it with a [simple version](https://github.com/ros2/examples/issues/43#issuecomment-128771626) but it compiled fine.

My proposal to get back to Aug 5th in the Windows Jenkins node was more to have something working by now, while we find what happen with this bug and the confirmation that RC is compiling the same code just fine.

dirk-thomas, 2015-08-08:
Finally I got it working with the Community version of Visual Studio 2015 (final release) and a patch to the generated OpenSplice code: http://ci.ros2.org/job/ros2_batch_ci_windows/133/

dirk-thomas, 2015-08-08:
Done. The windows slave is ready to use again.

closed 2015-08-08 by dirk-thomas