ros2/ros2 #73: Pass timeout duration to rmw_wait

jacquelinekay, 2015-07-24:
https://github.com/ros2/rclcpp/pull/66
https://github.com/ros2/rmw/pull/25
https://github.com/ros2/rmw_opensplice/pull/55
https://github.com/ros2/rmw_connext/pull/60

jacquelinekay, 2015-07-27:
First CI:
http://ci.ros2.org/job/ros2_batch_ci_linux/93/
http://ci.ros2.org/job/ros2_batch_ci_osx/35/
http://ci.ros2.org/job/ros2_batch_ci_windows/62/

I will have to run the jobs again because these were launched before I actually added the test as a nosetest in CMakeLists.

jacquelinekay, 2015-07-29:
New CI:

New test fails on OSX:
http://ci.ros2.org/job/ros2_batch_ci_osx/36/

Linux passes:
http://ci.ros2.org/job/ros2_batch_ci_linux/94/

Windows builds and many tests fail (consistent with master):
http://ci.ros2.org/job/ros2_batch_ci_windows/67/

I've moved this back to "In progress" while I debug the OSX timeout issue.

jacquelinekay, 2015-07-29:
new OSX job passes after I increased the tolerance by 5 ms:

http://ci.ros2.org/job/ros2_batch_ci_osx/48/

The tolerance is now 15 ms, for a timeout of 100 ms.

dirk-thomas, 2015-07-29:
Please rebase your PRs to the latest master and run the jobs again to ensure that there are no regressions.

jacquelinekay, 2015-07-29:
I rebased and started CI. But as you know, the Linux job is currently disabled, so that will have to wait.

http://ci.ros2.org/job/ros2_batch_ci_osx/50/
http://ci.ros2.org/job/ros2_batch_ci_windows/

dirk-thomas, 2015-07-29:
Here is the Linux job:

http://ci.ros2.org/job/ros2_batch_ci_linux/109/

dirk-thomas, 2015-07-30:
The referenced jobs indicate new warnings related to these PRs. Can you please fix them and post new job links?

jacquelinekay, 2015-07-31:
The  newwarnings appear to be due to implicit type conversion. Launched new CI with the fix (I couldn't test it locally because it's on Windows):
http://ci.ros2.org/job/ros2_batch_ci_linux/118/
http://ci.ros2.org/job/ros2_batch_ci_osx/60/
http://ci.ros2.org/job/ros2_batch_ci_windows/87/

jacquelinekay, 2015-07-31:
Some of the warnings are due to MSBuild giving warnings about implicit type conversions from unsigned int to signed int, so I couldn't test them locally.

New CI:
http://ci.ros2.org/job/ros2_batch_ci_linux/118/
http://ci.ros2.org/job/ros2_batch_ci_osx/60/
http://ci.ros2.org/job/ros2_batch_ci_windows/87/

dirk-thomas, 2015-07-31:
The warnings referenced in the last two links could be the case if your branch is not based on a current master.

jacquelinekay, 2015-07-31:
yep, had to rebase rmw_connext and system_tests... here we go again:

http://ci.ros2.org/job/ros2_batch_ci_windows/91/

dirk-thomas, 2015-08-04:
This needs another round of CI jobs after rebasing since some tests failed.

closed 2015-08-05 by jacquelinekay

dirk-thomas, 2015-08-10:
The ros1_bridge package (https://github.com/ros2/ros1_bridge) needs the same updates to continue to compile. I will create a patch for it since I need to compile the package for ros2/ros1_bridge#3.