ros2/ros2 #74: Prevent triggering debugger on Windows slave

dirk-thomas, 2015-07-28:
So far we have tried a few things without success: http://stackoverflow.com/questions/396369/how-do-i-disable-the-debug-close-application-dialog-on-windows-vista/554596

j-rivero, 2015-08-03:
Do we have an easy way of manually triggering the problem?

dirk-thomas, 2015-08-03:
Every job (http://ci.ros2.org/job/ros2_batch_ci_windows/) will trigger the Debugger when the test fail at the end.

j-rivero, 2015-08-05:
In my first exploration into the windows CI machine, I found that the firewall window was in foreground and asking for permission to run some tests. That could make test to fail. For the gazebo testing machine I've disable the firewall completely to be able to run tests that require some connection with the network. I assume that I can do the same for ROS machine, right?

Just to confirm, your real problem is the debug window and not the firewall, isn't it?

dirk-thomas, 2015-08-05:
Yes, you can disable the firewall.

Yes, the real problem is that the debugger triggers and therefore every failing test is waiting until the timeout.

j-rivero, 2015-08-10:
Seems like the slave is back to that state of no desktop manager. Can we configure an auto-login so I can disable jenkins + reboot and back to connect to it? Thanks.

wjwwood, 2015-08-11:
Uh, sure let me set that up for you.

wjwwood, 2015-08-11:
Ok, auto login seems to be setup now.

j-rivero, 2015-08-11:
Working fine, thanks!

j-rivero, 2015-08-14:
My investigation (Windows 7 system) on stopping the debugging window:

 * The [recommend way](https://msdn.microsoft.com/en-us/library/e631wekh.aspx) I found in all pages is to include the code a couple of lines to disable the debugger Window. We would need to include the code in every executable which is not nice. The most complete text describing text I found with the solution is: http://www.zachburlingame.com/2011/04/silently-terminate-on-abortunhandled-exception-in-windows/
 * 'stracing' an abort test case with Process Monitor (cool tool), discovered that: cmd.exe -> svchost.exe -> WerFault.exe (Windows Error Reporting).
 *  If I remove the `WerFault.exe` or change it by a dummy stub returning success, Windows display an internal exception problem Window when I call my abort executable.
 * If I use `msconfig` to disable the service `Windows Error Reporting` the debbuger window is still being displayed.
 * What worked for me was to modify the [Wer Setting in the registry](https://msdn.microsoft.com/es-es/library/windows/desktop/bb513638%28v=vs.85%29.aspx), `HKEY_CURRENT_USER\Software\Microsoft\Windows\Windows Error Reporting` and set `Disabled` and `DontShowGUI` to 1.

When I tried to reproduce the failure in our Jenkins node, my minimal cpp file calling abort did not trigger the debugger window ... and when I looked into the registry, the Disabled and DontShowGUI keys where set to 1. Did anyone change it? Is the problem now solved or jenkins is still triggering the Debugger?

dirk-thomas, 2015-08-14:
I set those values but for the failing tests it still shows the debugger dialog.

tfoote, 2018-03-08:
I believe we resolved this a while ago.

closed 2018-03-08 by tfoote