ros2/ros2 #78: should the CI jobs automatically rebase to master

dirk-thomas, 2015-07-30:
Currently the CI job tests the code from the specified branches. If they are not up-to-date but branches earlier and master has diverged that is not covered by the test. The result is that the CI job might succeed but after merging the branch the CI job might fail on master.

Would it be sufficient to always do this after switching to the to-be-tested branches:

    vcs custom --git --args rebase master

jacquelinekay, 2015-07-31:
+1 to this idea, with a few questions about a few details:

1. How should the CI job continue if there are merge conflicts in the rebase? I think it should abort the rebase and continuing testing the branch with a warning to the user.
2. Should the CI job `push --force` the rebase if the rebase has no conflicts? Should it wait to push the rebase until after the CI job succeeds?? I think it should be the first option, since we will always want to rebase to master eventually.

esteve, 2015-07-31:
Sounds good. In the case where a conflict happens, I think it's useful to still run the CI job, but warn the user, though ultimately GitHub will complain the branch not being mergeable.

I have mixed feelings about `push --force` the rebase, while it's useful, I rather not overwrite anyone's branches.

dirk-thomas, 2015-07-31:
+0.5 to build anyway if rebasing fails and place a warning somehow.
+1 for *not* pushing to the repositories (ever).

tfoote, 2015-07-31:
I agree we shouldn't let the CI push to the repos. 

For falling back to a build without a rebase, i'd suggest that be an optional parameter. By default a pass should be that it merged with master and passed unit tests. 

There are cases when you will want to test a diverged branch on the farm, but in that case you should know they've divered and tell it, don't merge/rebase.

wjwwood, 2015-08-02:
+1 to merge before build
+0 to build if merge fails (travis doesn't do this; I don't think we need to either)

PR #87 (2015-08-17): rebase all repos to master which must succeed
Fixes #78.
closed 2015-08-17 by dirk-thomas