ros2/ros2 #79: Are ROS2 nodes hot swappable?

ruffsl, 2015-07-31:
# Issue
I've been messing about with the ROS2 talker and listener demos and have encountered a behavior leading me to ask if ROS2 nodes are hot swappable? What I've done so far is create two networks, and two pairs of containers running talker and listener demo, one pair attached to each network. I then witness the auto discovery has successfully connected the pairs of nodes, i.e. the listers are echoing the messages from their paired talker. Afterwards I detach and then attach the talker containers from their original networks and swap them in pace, then wait for the listers to resume echoing the message topic.

# Expected
I expected the reorganized pair of ROS2 nodes to again perform auto discovery, with the subscriber eventually resuming to echo messages from the new publisher.

# Result
Node listener instead remain silent even well after a minute (60 sec being the longest time I've yet seen an auto discovery to successfully complete). I then forfeited the experiment and swapped the talker containers back to there original network. With in seconds I noted that the listeners had resume their function.

# Reproduce
The procedure requires Docker v1.8 for networking capability.

> First create two networks with docker:

    docker network create foo
    docker network create bar

> Then spin up a talker and listener pair attached to a common network. Launching the node and attaching it to the network can be done like so:

    docker run -it --rm \
        --publish-service=talker.foo \
        --name talker1 \
        osrf/ros2 \
        /root/ros2_ws//install/bin/talker

> In a separate terminal on the same host, launch the listener:

    docker run -it --rm \
        --publish-service=listener.foo \
        --name listener1 \
        osrf/ros2 \
        /root/ros2_ws//install/bin/listener

> Again, do this same for the the talker on the other network:

    docker run -it --rm\
        --publish-service=talker.bar \
        --name talker2 \
        osrf/ros2 \
        /root/ros2_ws//install/bin/talker

> As well as for the other listener:

    docker run -it --rm\
          --publish-service=listener.bar \
          --name listener2 \
          osrf/ros2 \
          /root/ros2_ws//install/bin/listener

You should see the pair of nodes are happily chatting away, with the number count in the message offset between the pair by the amount of time it took you to start the second talker after the first.

> The current structure looks as so:

    $ docker service ls
    SERVICE ID          NAME                NETWORK             CONTAINER
    9042e7ba1917        talker              bar                 503e46757b40
    ed18095004ad        listener            bar                 59fcfe9a59cb
    44ba5d512225        talker              foo                 1a4f8d4a36ab
    8946add1898b        listener            foo                 9768dcb22729

> Now lets detach the two talkers from there original network

    docker service detach talker1 talker.foo
    docker service detach talker2 talker.bar

> The talker nodes will then begin to networking error message looking like so:

```
========================================================================================
Report      : ERROR
Date        : Fri Jul 31 18:08:50 UTC 2015
Description : ddsi_conn_write failed -1

Node        : 1a4f8d4a36ab
Process     : talker <1>
Thread      : tev 7f5e55879700
Internals   : V6.4.140407OSS///ddsi2/q_xmsg.c/1051/0/1438366130.827112079
```

> Listeners will also message about a missing hearbeat from thier lost companion

```
========================================================================================
Report      : WARNING
Date        : Fri Jul 31 18:05:37 UTC 2015
Description : Missed heartbeat for node - heartbeat id : 465741851. (DCPSPublication state 1)
Node        : 9768dcb22729
Process     : listener <1>
Thread      : OSPL Garbage Collector 7ff06f6ea700
Internals   : V6.4.140407OSS///v_spliced/v_spliced.c/2439/0/1438365937.347569575
```

> Now then complete the swap of the container by reattaching them to the other network:

    docker service attach talker1 talker.bar
    docker service attach talker2 talker.foo

> You should now have a structure that looks like so, notice the two container IDs for the talker services have now swapped:

    $ docker service ls
    SERVICE ID          NAME                NETWORK             CONTAINER
    44ba5d512225        talker              foo                 503e46757b40
    8946add1898b        listener            foo                 9768dcb22729
    9042e7ba1917        talker              bar                 1a4f8d4a36ab
    ed18095004ad        listener            bar                 59fcfe9a59cb

> The talkers will fail to find their new partner. If we then revers our network changes we can return the system to working order:

    docker service detach talker1 talker.bar
    docker service detach talker2 talker.foo
    docker service attach talker1 talker.foo
    docker service attach talker2 talker.bar

dirk-thomas, 2015-07-31:
I don't this is related to any ROS specific part but solely depends on the DDS vendor behavior. I would suggest looking into the network detail, e.g. if network interfaces are replaces by Docker and if DDS is aware of that.

wjwwood, 2015-11-04:
I'm seeing `ddsi_conn_write failed -1` on my Mac as well. I think it might be a common error which can be causes be several different things.

wjwwood, 2015-11-04:
We found that `ddsi_conn_write failed -1` on OS X was caused by having too small of a max UDP packet size. But it could really be any system failure when doing a write and OpenSplice doesn't do a good job of reporting the `errno` and the error string.

ruffsl, 2018-02-12:
Just curious so I looked back at this issue again now using ardent. Updating the example for current docker via `docker-compose.yml` file in a folder named `demo`, I still encounter the same issue where talker and listener pairs, when swapped networks, will not connect with the other node on the same network.

### `docker-compose.yml`

``` yaml
version: "3.3"

services:
  talker.foo:
    image: osrf/ros2:ardent-basic
    command: ros2 run demo_nodes_cpp talker
    networks:
      - foo

  listener.foo:
    image: osrf/ros2:ardent-basic
    command: ros2 run demo_nodes_cpp listener
    networks:
      - foo

  talker.bar:
    image: osrf/ros2:ardent-basic
    command: ros2 run demo_nodes_cpp talker
    networks:
      - bar

  listener.bar:
    image: osrf/ros2:ardent-basic
    command: ros2 run demo_nodes_cpp listener
    networks:
      - bar

networks:
  foo:
  bar:
``` 

### Commands

``` shel
cd demo
docker-compose up

# From wew shell, disconnect talkers or listeners
docker network disconnect demo_foo demo_talker.foo_1
docker network disconnect demo_bar demo_talker.bar_1

# Swap them by re-connecting talkers or listeners to opposite network
docker network connect      demo_foo demo_talker.bar_1
docker network connect      demo_bar demo_talker.foo_1

# Note that neither listener prints chatter topics until swap is reverted
docker network disconnect demo_foo demo_talker.bar_1
docker network disconnect demo_bar demo_talker.foo_1
docker network connect      demo_foo demo_talker.foo_1
docker network connect      demo_bar demo_talker.bar_1
```