ros2/ros2 #86: Add support for loading QoS profiles via XML files

esteve, 2015-08-17:


jacquelinekay, 2015-08-18:
It looks like the premise of these changes is under discussion in ros2/rclcpp/pull/92.

Ignoring that: perhaps you could add a test for QoS profile parsing?

tfoote, 2015-09-15:
We agreed not to support the RTI XML format, but to support some form of the future standard DDS format, but not through the API.

ruffsl, 2018-07-13:
Is there a current workaround for to update QoS parameter at runtime with ROS2? Even if vendor specific?

<details>

I'm attempting to control `dds.transport.shmem.builtin.host_id` and `rtps_app_id` QoS parameters to better configure shared-memory IPC over containers: [Communicate two Docker containers using RTI Connext DDS and shared memory](https://community.rti.com/kb/communicate-two-docker-containers-using-rti-connext-dds-and-shared-memory)

mikaelarguedas, 2018-07-13:
FastRTPS loads XmlProfile automatically at node creation: https://github.com/ros2/rmw_fastrtps/blob/7d3ff15e40999d878fb67104987b3d654d48db17/rmw_fastrtps_cpp/src/rmw_node.cpp#L237-L238

I don't think the equivalent is available for Connext since they were using their own XML profile files

ruffsl, 2018-09-11:
I created a question about this for Connext:  
https://answers.ros.org/question/303016/setting-qos-with-ros2-using-rmw_connext_cpp/

closed 2023-07-24 by clalancette