ros2/ros2 #91: Implement subscription side filtering when intra process is being used

wjwwood, 2015-08-19:
Currently, there is nothing stopping a subscriber from receiving messages from both the intra process mechanism and the inter-process mechanism. Therefore publishers and subscriptions connected by intra process should avoid (through filtering or some other mechanism) exchanging messages using the inter process communications.

So far I have looked into:

- Publisher side filtering: before calling rmw_publish, make sure someone is going to get it, i.e. `len(connected_subscritpions - intra_process_subscriptions) > 0`.
 - The problem with this is latching. If we never call rmw_publish then when a new subscription comes up the message will not be delivered via the latched mechanism.
- Using the DDS functions like `ignore_subscription` (https://community.rti.com/static/documentation/connext-dds/5.2.0/doc/api/connext_dds/api_cpp/classDDSDomainParticipant.html#ad9b1a187eabf36e2c370bbe1b87f76b0)
 - I haven't gotten this to work as expected, I think because of when the function must be called (cannot be called after creating the publisher), but it needs more investigation.
 - It is not implemented by some vendors like OpenSplice.
- Subscription side filtering: simply not delivering messages received from inter process from publishers which are connected via intra process.
 - This is inefficient since the publisher has to serialize and send and the subscription must then receive and deserialize the message before dropping it.
 - Without help from the middleware this may have to happen anyways (like if multicast is being used).

As a stop gap measure we should implement the subscription side filtering so that users don't get duplicate callbacks when intra process is being used.

If other methods become available then we can change how we do it then.

jacquelinekay, 2015-08-19:
I like the idea behind the first approach best, but it may mean creating a hack-y code path for latched messages (if publishing a latched topic, store an "in flight" message in the intra-process manager,  and short-circuit `take_intra_process_message` for the subscriber to the latched topic when it instantiated).

wjwwood, 2015-08-19:
@jacquelinekay suggested we test:

- Test multiple nodes doing intra process.
- Test a case where the publisher has intra process enabled but the subscription doesn't, and vice versa.

wjwwood, 2015-08-19:
> I like the idea behind the first approach best, but it may mean creating a hack-y code path for latched messages (if publishing a latched topic, store an "in flight" message in the intra-process [...]

Yeah, the issue is with inter process subscriptions. If we never call rmw_publish (in order to avoid duplicate delivery to the intra process subscription) then we cannot rely on DDS's persistence QoS to handle latching for us (because we never gave it the message to persist). It's likely that the first solution would also need the subscription filtering in order to be used with latching.

wjwwood, 2015-08-25:
Ok, I've now tested it against opensplice and it works. I still don't have a good way to prove it in a test. I tested it by printing messages when entering each handle function (handle intra and handle inter processing messages) as well as printing something when a inter process message is explicitly dropped.

I need to make some changes to the rclcpp API to really test end to end pointer preservation. That will probably be the most convincing proof that intra process is working. That coupled with asserting that no duplicate messages get through should be a good test.

Also based on @dirk-thomas's suggestion take now optionally returns a `rmw_message_info_t` struct which currently only contains a `rmw_gid_t` for the publisher, but could later include things like the received timestamp and maybe other stuff.

wjwwood, 2015-08-25:
I still haven't gotten to the tests that @jacquelinekay suggested (I mention them higher in this thread).

I think we should get the outstanding pull requests merged and then I can open more pull requests to adjust the rclcpp API to include more callback signatures for testing and to add the missing tests.

wjwwood, 2015-08-25:
Before merge I have to implement for Connext and Connext Dynamic. I'll do that now and report back when it's ready for final review.

wjwwood, 2015-08-25:
Ok, Connext and Connext Dyanmic are implemented. I'm going to write another system test or two and then this will be ready for final review.

wjwwood, 2015-08-25:
I'm still working on a system test, but here are some preliminary CI jobs:

- http://ci.ros2.org/job/ros2_batch_ci_linux/270/
- http://ci.ros2.org/job/ros2_batch_ci_osx/190/
- http://ci.ros2.org/job/ros2_batch_ci_windows/270/

wjwwood, 2015-08-26:
I fixed some tests and addressed outstanding uncrustify style stuff, here are new CI:

- http://ci.ros2.org/job/ros2_batch_ci_linux/271/
- http://ci.ros2.org/job/ros2_batch_ci_osx/191/
- http://ci.ros2.org/job/ros2_batch_ci_windows/271/

This is ready for review. I'll squash after review.

Now I'm working on the ability to receive the message info object in the rclcpp callback so that we can write tests which assert that intra process is being used. That PR will depend on this set. Once I have that I can write the tests described above.

jacquelinekay, 2015-08-26:
Everything looks good except for the Windows build. What's going on there?

wjwwood, 2015-08-26:
@jacquelinekay I have no idea, I was hoping it would just go away. I'm looking into it now...

wjwwood, 2015-08-26:
I guess I need to turn on `/bigobj`: https://msdn.microsoft.com/en-us/library/8578y171.aspx

Either that or start refactoring our code base...

wjwwood, 2015-08-26:
Ok, I added that flag, we'll see if it fixes the problem for now, new CI:

- http://ci.ros2.org/job/ros2_batch_ci_linux/273/
- http://ci.ros2.org/job/ros2_batch_ci_osx/193/
- http://ci.ros2.org/job/ros2_batch_ci_windows/273/

wjwwood, 2015-08-26:
Try, try again: http://ci.ros2.org/job/ros2_batch_ci_windows/274/

wjwwood, 2015-08-26:
This one should fix the warnings:

- http://ci.ros2.org/job/ros2_batch_ci_windows/275/

I'll also run the other CI again to ensure it didn't break them:

- http://ci.ros2.org/job/ros2_batch_ci_linux/274/
- http://ci.ros2.org/job/ros2_batch_ci_osx/194/

wjwwood, 2015-08-26:
The warnings have been fixed on Windows and the other two jobs are still good.

There is a new failing test on OS X, but I don't think it is related, and instead might be flaky. I don't want to take the time to debug that right now, so I opened an issue about it: https://github.com/ros2/system_tests/issues/36

There is also a new failing test on Linux (different from the one on OS X) which I saw while writing this. I'll open an issue for that too. It also seems unrelated to these changes and it passed on the previous build.

wjwwood, 2015-08-26:
Follow up issue: https://github.com/ros2/system_tests/issues/37

wjwwood, 2015-08-26:
I think the outstanding pr's are ready to merge.

closed 2015-08-26 by wjwwood