ros2/ros2 #92: Audit intra process comms for thread safety

wjwwood, 2015-08-19:
At best implement tests which exercise the thread safety of the intra process manager class, but also audit the design considering the possible ways in which lighting may strike.

This is a follow on to https://github.com/ros2/ros2/issues/59

jacquelinekay, 2015-08-25:
I made the demo single-threaded for now, to avoid possible lightning strikes (ros2/demos/pull/12).

tfoote, 2015-12-01:
@gerkey to take a look.

jacquelinekay, 2015-12-03:
@wjwwood, do you consider this ticket to be resolved by recent improvements to fix race conditions? I don't believe that we've fixed all multithreading issues, but I'm fine with closing this issue for now and ticketing new, more specific issues about thread safety in the future.

wjwwood, 2015-12-03:
That's ok with me, this ticket was just there to remind us that I didn't consider thread-safety closely when originally writing it.

closed 2015-12-03 by wjwwood