ros2/ros2 #93: Add system default QoS profile

esteve, 2015-08-19:


esteve, 2015-08-19:
CI jobs:

http://ci.ros2.org/job/ros2_batch_ci_linux/225/
http://ci.ros2.org/job/ros2_batch_ci_windows/228/ (not yet started)
http://ci.ros2.org/job/ros2_batch_ci_osx/148/ (not yet started)

dirk-thomas, 2015-08-19:
The patches look ok to me.

I think we need to catch up on ros2/design#36 to describe the big picture and use cases and from that derive the options we want to expose and how the user can override the them. Currently with the incremental changes it is a bit unclear to me.

esteve, 2015-08-19:
@dirk-thomas sure. These patches just let the user leave up to the underlying implementation to pick whatever QoS values are appropriate. For example, if using RTI Connext, the user can have an XML file with QoS profiles, and RTI Connext will automatically load it if it's in `$NDDSHOME/resource/qos_profiles_5.2.0/xm/NDDS_QOS_PROFILES.xml`

In any case, we already agreed to update the design document once the QoS infrastructure is in a more complete shape.

jacquelinekay, 2015-08-20:
+1 to these patches

closed 2015-08-21 by esteve