ros2/ros2 #95: Add codepath for locking the size of dynamically sized messages

jacquelinekay, 2015-08-20:
Suggestion: during the initialization phase of node lifecycle, provide a way of limiting the size of dynamic messages objects so that they can be preallocated before node execution.

Original posts from @adolfo-rt in https://github.com/ros2/demos/pull/7:

"What I'd like to see in the long run is the ability to preallocate messages in publishers and subscribers, so dynamically-sized fields can be used. Something like:

```
sensor_msgs::msg::JointState msg;
msg.name.push_back("pendulum");
msg.position.resize(1, 0.0);
msg.velocity.resize(1, 0.0);
msg.effort.resize(1, 0.0);

publisher.preallocate(msg);  // optional
subscriber.preallocate(msg); //
```

preallocate would initialize with msg all internal buffers in the non real-time path. When publishing and subscribing from the real-time path, no further allocations are required as long as the vector and string sizes remain the same.

I think it's reasonable to assume that message and other data structure sizes can be determined (and pre-allocated) at initialization/configure time, which is non real-time. The active/running state would incur no further allocations. I find it also reasonable to expect a node reconfiguration (something like stop→cleanup→configure→start) to account for a problem size change.

I don't doubt there are very legitimate usecases out there for hard-coded message upper bounds, but I've so far not had the need for them. On the other hand, I can't imagine generic robot control software without (pre-allocated) dynamically sized messages."

Pros:
Allows more flexibility for using dynamic messages in a real-time fashion (previously we only considered two kinds of messages viable for real-time: statically sized messages and dynamically sized messages with an upper bound specified in IDL).

Cons:
If a publisher has put a limit on the size of a message but the subscriber publishes a message larger than that limit, the publisher will potentially lose data.

dirk-thomas, 2015-08-20:
I don't see how that should be possible for the DDS internal part. The generated code for DDS messages can't be altered at runtime and these messages are unbounded.

adolfo-rt, 2015-08-20:
> Cons:
If a publisher has put a limit on the size of a message but the subscriber publishes a message larger than that limit, the publisher will potentially lose data.

Instead of losing data, I would prefer a `resize()` to take place. I think it's better to incur once in a (potential) violation of real-time constraints, than to start losing data from this moment on. If you have a hard real-time system, the violation will be detected by 'the system' and a system failure would be triggered. Also, note that typically `container.resize(container.size())` is a no-op.

I don't know the extent to which this applies to DDS.

> I don't see how that should be possible for the DDS internal part. The generated code for DDS messages can't be altered at runtime and these messages are unbounded.

One can already publish messages with dynamically-sized payloads in non real-time. What's missing is the ability to programatically specify the size of buffers to preallocate, at runtime, not IDL writing time. Is this not possible with DDS (I really hope it is)?. If not, then at least we have a strong selling point for a non-DDS intra-process transport.

jacquelinekay, 2015-08-20:
The immediate benefit of this feature would be allowing messages with string fields to be used in real-time execution, which opens up most of the standard message definitions. We currently do not have a way of declaring a string message with a bounded upper size (maybe a bounded array of chars? ugh).

There is also the added benefit of reusing existing dynamically-sized, unbounded messages in generic tools: for example, an image publisher that figures out the desired size of messages at runtime and allocates exactly a 640x480 buffer for its message data, or a 1024x768 buffer, etc.

On limiting the size of DDS objects:

I think this would require the bounds specified at the `rcl` layer to be passed to `rmw` so that bounds can be enforced on the DDS data structures. It would create a lot more complexity and work to be done in the rmw implementations. We can do things like setting a limit on the maximum sample size in `take` and preallocating `SampleInfoSeq` and `DataSeq` objects, rather than instantiating them before `take` is called (see https://github.com/ros2/rmw_opensplice/blob/master/rmw_opensplice_cpp/src/functions.cpp#L132). I'm pretty sure that instantiating a Seq object incurs dynamic allocations...

Beyond how we use the API in the `rmw` implementation, allocations made by the underlying DDS implementation are beyond our control. It's up to RTI/Prismtech/whoever to make guarantees about the real-time performance of their DDS implementation.

dirk-thomas, 2015-08-20:
@jacquelinekay We do have the ability to specify the upper boundary of strings.

jacquelinekay, 2015-08-20:
yeah, I forgot that we do have a way of specifying the upper boundary of strings. The issue I was thinking of is that the C++ struct generated by `rosidl` for a bounded string field is no different from the generated code for an unbounded string, as you can see in https://github.com/ros2/rosidl/blob/master/rosidl_generator_cpp/resource/msg__struct.hpp.template

adolfo-rt, 2015-08-20:
> The immediate benefit of this feature would be allowing messages with string fields to be used in real-time execution, which opens up most of the standard message definitions.

Not only strings. Vectors of floating-point numbers and other numeric structures (think a PID gains message) are commonplace in control code. A general-purpose control message will specify these as dynamically-sized vectors, so it can work on arbitrary-size robots (e.g., a few joints or many).

BobDean, 2015-09-12:
DDS deals with this by internally defining a max bound for unbounded sequences. Or at least RTI did when i asked gordon about this 3 years ago. Within our code we have rframe::FixedString<N> class, which provides std::string interface, but on a preallocated buffer, and a DynamicArray<N> class which provides bounded array capabilities.  Serialization is then smart enough to only send size() entries instead of max_size().

dirk-thomas, 2015-09-12:
This is not necessarily the case with Connext 5.2 since it finally allows really unlimited bounds. So if a ROS message is defined with unbounded arrays there are also truely unbounded in Connext.

The default upper bounds in Connext (when not using --unbounded*) are really small (100 for sequences, 255 for strings) which was not realistic for many messages anyway.

BobDean, 2015-09-12:
@dirk-thomas , are these unbounded arrays supported within their custom components such as sql and excel logging? my overcome-by-events conversation was centered on how data classes were unrollled into sql, which required predefined column names.

dirk-thomas, 2015-09-12:
I have never used any of these Connext parts so I don't know.