ros2/ros2 #106: Jenkins test result reporting

dirk-thomas, 2015-08-31:
We currently use:
* the xUnit plugin, configured for GoogleTest result files
  * the problem is that is does not detect tests with an "error", e.g. generated by nosetests

Before we used:
* the xUnit plugin, configured for JUnit result files
  * the problem was that it is unable to parse `gtest` result files (#77)

If I remember correctly:
* the JUnit plugin was not able to parse all result files either

dirk-thomas, 2015-09-02:
Since I don't see a way to use a single Jenkins plugin / configuration to handle all different formats correctly I:
* added the test result file type as a second extension (e.g. `.gtest.xml`, `.xunit.xml`)
* generate the CTash dashboard summary to catch test which are e.g. label `Not Run`

https://github.com/ament/ament_cmake/commit/ff6da309cfcf8d721b69e01fb462594a7ef58e5f
https://github.com/ament/ament_lint/pull/27
https://github.com/ament/ament_tools/pull/48

The job configuration is changed to handle the three different result file type (CTest, GTest, XUnit) separately: https://github.com/ros2/ros2/commit/cb12b127c03a72b7343c56ff7ff8d4155b69efa1

This will inadvertently report most failing tests twice (CTest + other category) but I don't see how to avoid that without loosing information about e.g. not run tests.

The Windows job already uses the new configuration:
http://ci.ros2.org/job/ros2_batch_ci_windows/360/
http://ci.ros2.org/job/ros2_batch_ci_windows/361/

dirk-thomas, 2015-09-04:
While it looks good right now I want to create a set of tests for this before closing it. Back to in-progress...

dirk-thomas, 2015-09-15:
I have create the repo https://github.com/ament/test_jenkins_test_results which contains gtests, nosetests as well as example code to be linted. For each of those there are succeeding as well as failing cases (also time outs if applicable).

The Jenkins view on those can be seen for the following three jobs:
* http://ci.ros2.org/job/ros2_batch_ci_linux/360/testReport/
* http://ci.ros2.org/job/ros2_batch_ci_osx/268/testReport/
* http://ci.ros2.org/job/ros2_batch_ci_windows/406/testReport/

Please take closer look at the tests since they contain valuable comments about the behavior of gtest and ament test:
* https://github.com/ament/test_jenkins_test_results/blob/master/test_ament_cmake_gtest_test_results/test/gtest_successful_test_result.cpp
* https://github.com/ament/test_jenkins_test_results/blob/master/test_ament_cmake_gtest_test_results/test/gtest_timeout_test_result.cpp

With the new repository containing the test packages as well as the references PR I consider this done.

dirk-thomas, 2015-09-16:
Please review this meta ticket in order to close it.

wjwwood, 2015-09-16:
The output looks fine to me, and appears to line up with what you put in the comments of the example files.

Is the intention to leave this package integrated into the test suite? I guess not and it's just for testing now and will be discarded later?

Did you delete your comment?

dirk-thomas, 2015-09-16:
I will keep the new repository https://github.com/ament/test_jenkins_test_results around for future testing but it is not part of the ros2.repos file so it won't be used by default.

Yes, I removed my comment asking for review to keep the ticket history short.

wjwwood, 2015-09-16:
> Yes, I removed my comment asking for review to keep the ticket history short.

Please don't, there's no harm to having another comment. I opened the issue and the comment I got in the email was not there, making me think the page was out of date or the wrong issue. There's no visibility for that.

closed 2015-09-16 by dirk-thomas