ros2/ros2 #107: list_parameters segfaults on OS X

wjwwood, 2015-09-01:
On OS X (using the alpha1 binary) the `list_parameters` executable segfaults:

```
william@farl /tmp/ros2-alpha1
% list_parameters
[1]    84883 segmentation fault  list_parameters

william@farl /tmp/ros2-alpha1
% lldb list_parameters
(lldb) target create "list_parameters"
Current executable set to 'list_parameters' (x86_64).
(lldb) r
Process 84896 launched: '/tmp/ros2-alpha1/ros2-osx/bin/list_parameters' (x86_64)
Process 84896 stopped
* thread #1: tid = 0x97b11c, 0x00007fff8d6755c1 libsystem_pthread.dylib`_pthread_mutex_lock + 23, queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=EXC_I386_GPFLT)
    frame #0: 0x00007fff8d6755c1 libsystem_pthread.dylib`_pthread_mutex_lock + 23
libsystem_pthread.dylib`_pthread_mutex_lock:
->  0x7fff8d6755c1 <+23>: cmpq   $0x4d555458, (%rbx)
    0x7fff8d6755c8 <+30>: je     0x7fff8d6755de            ; <+52>
    0x7fff8d6755ca <+32>: movq   %rbx, %rdi
    0x7fff8d6755cd <+35>: callq  0x7fff8d6758c5            ; _pthread_mutex_check_init
(lldb) bt
* thread #1: tid = 0x97b11c, 0x00007fff8d6755c1 libsystem_pthread.dylib`_pthread_mutex_lock + 23, queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=EXC_I386_GPFLT)
  * frame #0: 0x00007fff8d6755c1 libsystem_pthread.dylib`_pthread_mutex_lock + 23
    frame #1: 0x00007fff86706b6f libc++.1.dylib`std::__1::mutex::lock() + 9
    frame #2: 0x000000010000a700 list_parameters`rclcpp::node::Node::set_parameters_atomically(std::__1::vector<rclcpp::parameter::ParameterVariant, std::__1::allocator<rclcpp::parameter::ParameterVariant> > const&) + 48
    frame #3: 0x000000010000a2af list_parameters`rclcpp::node::Node::set_parameters(std::__1::vector<rclcpp::parameter::ParameterVariant, std::__1::allocator<rclcpp::parameter::ParameterVariant> > const&) + 479
    frame #4: 0x000000010001d561 list_parameters`rclcpp::parameter_service::ParameterService::ParameterService(std::__1::shared_ptr<rclcpp::node::Node>)::'lambda'(std::__1::shared_ptr<rmw_request_id_t>, std::__1::shared_ptr<rcl_interfaces::srv::SetParameters_Request_<std::__1::allocator<void> > >, std::__1::shared_ptr<rcl_interfaces::srv::SetParameters_Response_<std::__1::allocator<void> > >)::operator()(std::__1::shared_ptr<rmw_request_id_t>, std::__1::shared_ptr<rcl_interfaces::srv::SetParameters_Request_<std::__1::allocator<void> > >, std::__1::shared_ptr<rcl_interfaces::srv::SetParameters_Response_<std::__1::allocator<void> > >) const + 449
    frame #5: 0x000000010001d307 list_parameters`std::__1::__function::__func<rclcpp::parameter_service::ParameterService::ParameterService(std::__1::shared_ptr<rclcpp::node::Node>)::'lambda'(std::__1::shared_ptr<rmw_request_id_t>, std::__1::shared_ptr<rcl_interfaces::srv::SetParameters_Request_<std::__1::allocator<void> > >, std::__1::shared_ptr<rcl_interfaces::srv::SetParameters_Response_<std::__1::allocator<void> > >), std::__1::allocator<rclcpp::parameter_service::ParameterService::ParameterService(std::__1::shared_ptr<rclcpp::node::Node>)::'lambda'(std::__1::shared_ptr<rmw_request_id_t>, std::__1::shared_ptr<rcl_interfaces::srv::SetParameters_Request_<std::__1::allocator<void> > >, std::__1::shared_ptr<rcl_interfaces::srv::SetParameters_Response_<std::__1::allocator<void> > >)>, void (std::__1::shared_ptr<rmw_request_id_t> const&, std::__1::shared_ptr<rcl_interfaces::srv::SetParameters_Request_<std::__1::allocator<void> > > const&, std::__1::shared_ptr<rcl_interfaces::srv::SetParameters_Response_<std::__1::allocator<void> > >&)>::operator()(std::__1::shared_ptr<rmw_request_id_t> const&, std::__1::shared_ptr<rcl_interfaces::srv::SetParameters_Request_<std::__1::allocator<void> > > const&, std::__1::shared_ptr<rcl_interfaces::srv::SetParameters_Response_<std::__1::allocator<void> > >&) + 135
    frame #6: 0x000000010001cd83 list_parameters`rclcpp::service::Service<rcl_interfaces::srv::SetParameters>::handle_request(std::__1::shared_ptr<void>&, std::__1::shared_ptr<void>&) + 147
    frame #7: 0x000000010002736b list_parameters`rclcpp::executor::Executor::execute_service(std::__1::shared_ptr<rclcpp::service::ServiceBase>) + 139
    frame #8: 0x0000000100027005 list_parameters`rclcpp::executor::Executor::execute_any_executable(std::__1::shared_ptr<rclcpp::executor::AnyExecutable>) + 261
    frame #9: 0x0000000100026cef list_parameters`void rclcpp::executor::Executor::spin_node_once<std::__1::ratio<1l, 1000l> >(std::__1::shared_ptr<rclcpp::node::Node>, std::__1::chrono::duration<long long, std::__1::ratio<1l, 1000l> >) + 143
    frame #10: 0x000000010002b530 list_parameters`std::__1::shared_future<std::__1::vector<rcl_interfaces::msg::SetParametersResult_<std::__1::allocator<void> >, std::__1::allocator<rcl_interfaces::msg::SetParametersResult_<std::__1::allocator<void> > > > >& rclcpp::executors::spin_node_until_future_complete<std::__1::vector<rcl_interfaces::msg::SetParametersResult_<std::__1::allocator<void> >, std::__1::allocator<rcl_interfaces::msg::SetParametersResult_<std::__1::allocator<void> > > > >(rclcpp::executor::Executor&, std::__1::shared_ptr<rclcpp::node::Node>, std::__1::shared_future<std::__1::vector<rcl_interfaces::msg::SetParametersResult_<std::__1::allocator<void> >, std::__1::allocator<rcl_interfaces::msg::SetParametersResult_<std::__1::allocator<void> > > > >&) + 80
    frame #11: 0x000000010000ec32 list_parameters`rclcpp::parameter_client::SyncParametersClient::set_parameters(std::__1::vector<rclcpp::parameter::ParameterVariant, std::__1::allocator<rclcpp::parameter::ParameterVariant> >) + 274
    frame #12: 0x000000010000cd50 list_parameters`main + 1040
    frame #13: 0x00007fff958575c9 libdyld.dylib`start + 1
    frame #14: 0x00007fff958575c9 libdyld.dylib`start + 1
```

wjwwood, 2015-09-01:
Possibly related, `list_parameters_async` fails with an exception:

```
william@farl /tmp/ros2-alpha1
% list_parameters_async
libc++abi.dylib: terminating with uncaught exception of type std::__1::system_error: mutex lock failed: Invalid argument
[1]    84911 abort      list_parameters_async
```

If they end up being unrelated I can spin off a new issue.

esteve, 2015-09-01:
Might be similar to the issue on Windows, though supposedly `std::mutex` is implemented the same way on OSX and Linux.

wjwwood, 2015-09-01:
`parameter_events` segfaults in the same way. We can look into it later. I'm just trying to catalogue the errors.

esteve, 2015-09-16:
@wjwwood does `list_parameters` work now with ros2/rclcpp#112 merged?

wjwwood, 2015-09-16:
Yeah, it runs now, but the output between the async and non async is different:

```
$ ./install/bin/list_parameters__rmw_opensplice_cpp
Parameter name: bar
Parameter name: foo
Parameter name: foo.first
Parameter name: foo.second
Parameter prefix: foo
william@farl:~/ros2
$ ./install/bin/list_parameters_async__rmw_opensplice_cpp
Parameter name: foo
Parameter name: foo.first
Parameter name: foo.second
Parameter prefix: foo
```

Not sure if that's expected or not.

closed 2015-09-16 by wjwwood

esteve, 2015-09-16:
It shouldn't, just filed a new ticket:

https://github.com/ros2/ros2/issues/121