ros2/ros2 #109: parameters crash on Node::set_parameters_atomically on Windows

esteve, 2015-09-01:
This might be caused by the way `std::mutex` is implemented on Windows (recursive mutex), look into converting that code into critical sections using the Windows API.

esteve, 2015-09-01:
Further investigation seems to indicate that the problem lies in using references to shared pointers, and it's not platform specific, though the issue doesn't seem to occur on Linux.

dirk-thomas, 2015-09-01:
Where do we still use references to shared pointers? I thought we removed those recently (#94).

esteve, 2015-09-01:
For example:

https://github.com/ros2/rclcpp/blob/master/rclcpp/include/rclcpp/service.hpp#L81
https://github.com/ros2/rclcpp/blob/master/rclcpp/include/rclcpp/parameter_service.hpp#L50

esteve, 2015-09-02:
#107 seems to be related to this

jacquelinekay, 2015-09-15:
Does ros2/rclcpp/#112 fix this and #107?

esteve, 2015-09-15:
@jacquelinekay it should. @wjwwood could you test locally? (i.e. via the examples)

wjwwood, 2015-09-15:
Sure, I'm building master now.

esteve, 2015-09-15:
@wjwwood thanks.

wjwwood, 2015-09-16:
lgtm

esteve, 2015-09-16:
Fixed via ros2/rclcpp#112

closed 2015-09-16 by esteve